dresden-elektronik/deconz-rest-plugin #495: Daylight sensor configuration

ebaauw, 2018-03-19:
Because the location discovery didn't quite set my exact coordinates, I updated `config.lat` and `config.long` manually, to the same values I use for the Hue bridge.  However, after that, the Daylight sensor's `state` is no longer updated.

I deleted the sensor (from the database) and restarted deCONZ.  It re-created the Daylight sensor, but it continued to show `state.configured` as false.  After another restart, the sensor is configured and its `state` is updated.

Upon inspecting the database, I see the automatic configuration sets `config.lat` and `config.long` to strings containing numeric values only, whereas the Hue API expects these strings to end with `N` or `S` or with `E` or `W`.  I did specify `N` and `E` when updating, and these did show in the database.

I think this is a bug?  Also, this makes me wonder how to coordinates on the southern or western hemisphere are handled.

ebaauw, 2018-03-19:
Oh, I also noticed that after restart, the Daylight sensor's `state.lastupdated` is off by an hour.  Normally it correctly shows the time in UTC, but after restart it shows UTC-1 (which matches my current timezone).  The database holds the UTC time alright.

manup, 2018-03-19:
In the current implementation only raw float values are expected, like they are provided for example in google and open street maps.

I'm not sure why hue bridge uses N, S, E and W since the coordinates already tell the exact location. We can strip them away in the API Request to be compatible with hue API.

manup, 2018-03-19:
>Oh, I also noticed that after restart, the Daylight sensor's state.lastupdated is off by an hour. Normally it correctly shows the time in UTC, but after restart it shows UTC-1 (which matches my current timezone). The database holds the UTC time alright.

Yes I've also notices that yesterday not sure what is causing this yet, maybe the current datetime parser treats the UTC date as local time.

closed 2018-06-12 by ebaauw