dresden-elektronik/deconz-rest-plugin #499: Atmos Edition- Xiaomi Aqara Temperature Humidity

ThomasWelzel, 2018-03-21:
Hello,

I installed the Xiaomi sensors today and the temperature and pressure values seem to be correct. 

But I don't know exactly what the humidity value is.

[image]
[image]
[image]
[image]

ThomasWelzel, 2018-03-21:
also the battery value is not there

ThomasWelzel, 2018-03-21:
[image]

ebaauw, 2018-03-21:
As per ZCL spec, humidity is in 0.01% (temperature in 0.01 degrees Celsius and pressure in hPa).  It might take up to an hour for the sensor reports its battery state.

ThomasWelzel, 2018-03-21:
But for boath sensors the 1 - x% is better than the  value for the humidity. The sended values are too Low.

closed 2019-03-30 by stale[bot]