dresden-elektronik/deconz-rest-plugin #501: Issues with 2.05.15 ?

erikproper, 2018-03-23:
Okay. It is in beta, so I guess I'm asking for trouble.
Notices two odd things in my case:
1: The device tab in the older WirelessLight web app remains empty in my case. I used to be able to at least see the Hue Tap's there. 
2: When I add Hue Tap's in the newer PWA app, I can initially edit the button assignments. However, after the initial opportunity, they are greyed out in the list when e.g. on http://office/pwa/index.html#control/group

Wait for the next version?

erikproper, 2018-03-23:
Just added two additional Taps (transferring them from the Phillips Hue bridge).
These ones are not greyed out yet, but ... the big "1" button doesn't switch off the lights on these two. The assigned lights that is.

erikproper, 2018-03-23:
Just discovered that after a reboot the Tap's are greyed out.

CDRX2, 2018-03-30:
Same here, my settings page on the older webapp remains empty. Just upgraded to 2.05.16, still the same.

erikproper, 2018-03-31:
Planning on testing it again in a few weeks. Haven't seen any responses from the developers though. But "good" to hear that I'm not the only one experiencing this ...

manup, 2018-03-31:
Confirmed, I'll have a look to it. From a quick look at some point the JS code on the devices page stops executing and the page ends up empty.

erikproper, 2018-03-31:
Note ... I listed two findings:

1: The device tab in the older WirelessLight web app remains empty in my case. I used to be able to at least see the Hue Tap's there.

==> Guess thats related to your "at some point the JS code on the devices page stops executing and the page ends up empty."

2: When I add Hue Tap's in the newer PWA app, I can initially edit the button assignments. However, after the initial opportunity, they are greyed out in the list when e.g. on http://office/pwa/index.html#control/group

==> This seems to be an other issue ..

manup, 2018-04-06:
With 2.05.19 the devices page should be visible again.

The Hue tap is still on the check list ..

CDRX2, 2018-04-06:
Thanks @manup, I updated last night and noticed the devices page is working again. Hadn't found the time to post yet...

erikproper, 2018-04-06:
Will be able to test Sunday / Monday. Just back from holidays.

-- 
|
| Prof.dr. Henderik A. Proper
|
| Luxembourg Institute of Science and Technology
| Luxembourg
|
| e.proper@acm.org
| +352-661425916
| www.list.lu
| www.ee-network.eu
|

erikproper, 2018-04-08:
Just upgraded to deconz-2.05.19-qt5.deb ... no luck. The XXXX/devices.html?v=2.05.19 remains empty ...

manup, 2018-04-12:
> Part #2 of the issue is still open. Or is this part of ongoing work: "The Hue tap is still on the check list .." ?

Yes this will be fixed in future versions.

erikproper, 2018-04-12:
Terrific! Will close the issue for now.

erikproper, 2018-04-12:
Terrific! Will close the issue now.

closed 2018-04-12 by erikproper