dresden-elektronik/deconz-rest-plugin #502: Zigbee-HAN supported?

mysteryan, 2018-03-23:
Just out of curiousity, anyone know if Zigbee HAN is supported in addition to HA/LL to interface to smart energy meters provided by municipalities?  Wondering if it can basically function like a Rainforest RAVen:  https://www.amazon.com/Rainforest-EAGLE-200-Energy-Monitoring-Gateway/dp/B07681Y7ZV/

roflmao, 2018-03-24:
Develco also produces some HAN interfaces, would these be supported?
https://www.develcoproducts.com/products/meter-interfaces/emi-norwegian-han/

ebaauw, 2018-03-24:
HAN is not a ZigBee protocol.  Reading the Develco documentation, it is the protocol/interface between a smart meter and a meter interface device, comparable to the end-consumer interface through the P1 port of the DSMR (Dutch Smart Meter Requirements).  It looks like there's also meter interface devices communicating with not so smart meters over IR, reading the pulse LED.

The meter interface device uses regular ZigBee HA to communicate with your home automation.  In other words: it's a gateway between HAN and ZigBee HA.  It provides _Simple Metering_ (0x0702) and _Electricical Measurement_ (0x0B04) clusters, also used by smart plugs (see #432).  deCONZ already exposes these as ZHAConsumption and ZHAPower sensors, so integrating electricity consumption into deCONZ should be easy.  For gas, water and heat (if these are exposed over HAN), we might need to provide different sensor types.

I checked out the Develco website earlier, searching for smart plugs.  Looking at the specs, they're by far the best I've encountered.  Also, I love the level of detail in their technical documentation.  However they only deliver to wholesale customers, not to consumers.  I haven't been able to find a distributor for the plug.  The meter interface device be delivered by the electricity network company or would you get one yourself as consumer?  How much do these devices cost?  I'm currently exposing my smart meter directly to HomeKit over the P1 port, see https://github.com/ebaauw/homebridge-p1, using a custom serial to USB cable which cost me € 20 (less when ordered in China, even less when soldering yourself, but I was feeling particularly lazy and impatient at that time :-).  I haven't seen any ZigBee-based solutions for DSMR.

Somewhat related: I suspect the so-called M-bus between "my" smart electricity and smart gas meters actually uses ZigBee Smart Energy.  This way, the P1 port on the electricity meter also provides information from the gas meter.  However, the meters are sealed and I have no access to this communication.

To proceed: please pair the meter interface device with deCONZ and post the usual deCONZ GUI screenshots here.

mysteryan, 2018-03-24:
When attempting to pair, it asks for an "install code" to join in a secure fashion.  I assumed this to be the TC Link Key, but was never successful pairing using the TC Link Key (with leading zeroes for EUI format).

ebaauw, 2018-03-25:
What are you trying to pair with what?  Who is asking for the install code?  From the description on Amazon, the Rainforest connects to the ZigBee network created by the utility company’s smart meter.  I think you’d get the install code from your utility company.

mysteryan, 2018-03-25:
I'm trying to connect to a Itron Centron smart meter: https://www.itron.com/na/technology/product-services-catalog/products/0/7/5/centron

They say the install code is typically found in the back of the device with the device id in EUI64 format or QR/bar code.  It requests this information from my utility's portal when trying to connect your own devices.  The device just needs to be ZigBee Smart Energy Profile 1.0 or 1.1 certified.  It was just a curiousity to see if it can interface to it and function like one of thise standalone units with all the recent commits relating to energy metering and didn't have any expectations it can easily be done.
[image]

ebaauw, 2018-03-25:
So you're initiating the connection from your utility company's portal?  And they ask for the device ID and install code of the device to be connected to the utility company's smart meter?

Maybe deCONZ can technically connect to the meter, but then you'd have to configure deCONZ as a router (in the _Network Preferences_ in the GUI), define an endpoint with a Smart Energy profile and the appropriate clusters (in the endpoints tab of that same window), and use the RaspBee's or ConBee's mac address as device ID.  I have no clue as to the security settings, whether Smart Energy uses a secret link key, or where to define the install code.  You would lose the network created by deCONZ (and the connectivity to your devices on that network).  Also, I doubt your utility company will let you connect to a device unknown to them or uncertified by them.

closed 2019-03-30 by stale[bot]