dresden-elektronik/deconz-rest-plugin #505: Sensor polling/attribute reporting

ebaauw, 2018-03-24:
Hey @manup, I'm still having some issues related to polling and/or attribute reporting of sensors where I could really use your help.

My first issue: Non-reportable attributes in a cluster with reportable attributes never get updated.

For the _Simple Metering_ cluster, the ubisys D1 supports attribute reporting for _Instantaneous Demand_, but not for _Current Summation Delivered_.  However, when the report for _Instantaneous Demand_ is processed by `updateSensorNode()`, `setZclValue()` seems to be called for every attribute (in general.xml?), not just for the attributes in the report (note the payload):
```
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:675 APS-DATA.indication srcAddr: 0x001fee0000001963, dstAddrMode: 2, profile: 0x0104, cluster: 0x0702, lqi: 255, rssi: 0
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:675 no button handler for: D1 (5503) cl: 0x0702 cmd: 0x0A pl[0]: 000
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:675 ZCL attribute report 0x001FEE0000001963 for cluster 0x0702, ep 0x04
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:675 	payload: 00042a0c0000
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:675 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0702
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:676 update ZCL value 0x0702/0x0000 for 0x001FEE0000001963 after 306 s
Mar 24 17:20:02 pi2 deCONZ[1492]: 17:19:57:676 update ZCL value 0x0702/0x0400 for 0x001FEE0000001963 after 306 s
```
This causes the `NodeValue`'s `timestampLastReport` to be updated, so the poll manager concludes all values are fresh enough and won't poll the cluster.  Consequently, _Current Summation Delivered_ never gets updated.

I'm facing a simimaler issue for the Heiman smart plug, which only seems to have 5 entries for it's reporting table, but six reportable attributes.  The last attribute that deCONZ tries to configure never gets updated.

I could of course not setup attribute reporting for the whole cluster, but that feels like a capitulation.  I think `updateSensorNode()` should only `updateZclValue()` for attributes in the event, but I'm not even sure if the event contains these.

My second issue: For smart plugs that don't support attribute reporting it makes sense to fast-poll the sensor node(s) when the plug is switched On or Off, just like polling the _Level_ and _Color_ clusters.  Maybe it might even be an idea to combine the polling of clusters in a device, irrespective of which nodes are used to expose these.  However, I don't think there's an object in deCONZ that represents a device and links to all it's nodes?

The polling of lights and sensors seems to be scheduled very differently.  On my test system (with only the Heiman connected), `PollManager::poll()` is called more often for sensors than for lights:
```
$ grep '>>>' /var/log/syslog.1 | sed 's/.*Poll //' | sort | uniq -c
   1607 light node Light 1
  33447 ZHAConsumption sensor node Consumption 2
  32740 ZHAPower sensor node Power 3
```
On my production system, however, `PollManager::poll()` is called less often for sensors:
```
$ grep '>>>' /var/log/syslog.1 | sed 's/.*Poll //' | sort | uniq -c
    666 light node Bathroom 1
    671 light node Bathroom 2
    665 light node Bathroom 3
    662 light node Bathroom 4
    674 light node Bathroom 5
    670 light node Bedroom 1
    669 light node Bedroom 2
    669 light node Bedroom 3
    671 light node Bedroom 4
    679 light node Dining Room 1
    678 light node Dining Room 2
    676 light node Dining Room 3
     51 light node Dining Room Table
    669 light node Downstairs 1
    678 light node Downstairs 2
    660 light node Entrance 1
    710 light node Hallway 1
    685 light node Jacco 1
    680 light node Jacco 2
    677 light node Jacco 3
    664 light node Katja 1
    621 light node Katja 2
    698 light node Kitchen Left
    994 light node Kitchen Right
    683 light node Kitchen Stove Left
    711 light node Kitchen Stove Right
    669 light node Living Room 1
    680 light node Living Room 2
    675 light node Living Room 3
    701 light node Living Room 4
    679 light node Living Room 5
    677 light node Living Room 6
    627 light node Living Room Desk
    692 light node Living Room Fan
    681 light node Living Room Left
    687 light node Living Room Right
    640 light node Terrace 1
    644 light node Toilet 1
    653 light node Trådfri 1
    519 light node Upstairs 1
    697 light node Upstairs 2
    651 light node Washroom 1
    223 ZHAConsumption sensor node Dining Room Table
    228 ZHAPower sensor node Dining Room Table
    480 ZHAPower sensor node Katja 2
    107 ZHAPower sensor node Living Room Desk
    204 ZHAPower sensor node Living Room Fan
    399 ZHASwitch sensor node Dining Room Switch Left
    104 ZHASwitch sensor node Dining Room Switch Right
```
I'm in the dark as to what causes the difference.  Note that `PollManager::poll()` gets called considerably less for the the ubisys D1 dimmer (Dining Room Table).  The dimmer does support attribute reporting, but so do the Trådfri lights (Trådfri 1 and Terrace 1) and the Osram Plugs (Living Room Desk, Living Room Fan, Katja 2).  Also note that currently all mains powered sensors are polled, including the ZHASwitch sensor nodes of the D1.

manup, 2018-03-24:
For the first issue I guess the referred commit should fix the problem.

> My second issue: For smart plugs that don't support attribute reporting it makes sense to fast-poll the sensor node(s) when the plug is switched On or Off, just like polling the Level and Color [...]

Currently this is all separated, should be okay with the upcoming changes to speedup light polling.

>The polling of lights and sensors seems to be scheduled very differently. On my test system (with only the Heiman connected), PollManager::poll() is called more often for sensors than for lights:

Yes I still have some uncommitted poll code here which speeds up polling of lights a lot, hope to get it ready in the next days.

ebaauw, 2018-03-25:
> For the first issue I guess the referred commit should fix the problem.

Thanks, @manup, that did the trick!  I put in the same code for the _Simple Metering_ cluster to deal with the Heiman.  My test environment is now back to polling the Heiman ZHAPower sensor every 3 seconds, since the polling doesn't update `timestampLastReport`, but it looks good on my production environment.

Here's a crazy thought: Could the same issue be happening for _Color Control_ for the OSRAM and Trådfri lights, with attribute reporting for _Current X_, _Current Y_, and _Color Temperature_, but not for _Enhanced Color Mode_?  Could that be the cause of #53?

closed 2018-06-12 by ebaauw