dresden-elektronik/deconz-rest-plugin #507: deCONZ is spamming my logfile

Krocko, 2018-03-25:
Good morning @manup. deCONZ writes every minute in the logfile.  Any chance to reduce this?

```
Mar 25 03:44:38 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:44:38 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:44:38 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:44:38 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:44:38 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:45:08 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:45:08 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:45:08 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:45:08 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:45:09 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:45:39 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:45:39 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:45:39 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:45:39 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:45:39 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:46:09 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:46:09 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:46:09 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:46:09 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:46:09 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:46:39 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:46:39 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:46:39 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:46:39 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:46:39 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:47:10 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:47:10 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:47:10 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:47:10 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:47:10 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:47:40 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:47:40 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:47:40 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:47:40 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:47:40 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
Mar 25 03:48:10 raspberrypi deCONZ-WIFI.sh[410]: scan available wifi
Mar 25 03:48:10 raspberrypi deCONZ-WIFI.sh[410]: command failed: Network is down (-100)
Mar 25 03:48:10 raspberrypi deCONZ-WIFI.sh[410]: check ip address
Mar 25 03:48:10 raspberrypi deCONZ-WIFI.sh[410]: no ip address for wlan0
Mar 25 03:48:10 raspberrypi deCONZ-WIFI.sh[410]: wifi not configured by deCONZ - cancel
```

manup, 2018-03-25:
Oops these are from debugging the WiFi script and will be silenced in the next releases, also the scanner should just run when the WiFI settings page fro Phoscon App is open.

Krocko, 2018-03-31:
Hy Manuel. Any news? Can i somehow deactivate deCONZ WIFI?

manup, 2018-03-31:
Only by manually editing the zll.db and set the wifi to deactivated, or wait a little bit until 2.05.18 is released &mdash; the noise will be removed there :)

Krocko, 2018-03-31:
I deactivated the wifi with the REST API. WIFI shows now "deactivated". But still the same.

[image]

manup, 2018-03-31:
Hmm damn, or you can try `$ systemctl stop deconz-wifi` for the running session.

Krocko, 2018-03-31:
Yes this works. Until a restart.
Will wait for 2.05.18

manup, 2018-04-06:
WiFi logger should be quite now since 2.05.19

Krocko, 2018-04-06:
Yes. Thank you.

closed 2018-04-06 by Krocko

Krocko, 2018-10-05:
@manup any help on this?

manup, 2018-10-05:
By the look of it some commands are missing the `>2 /dev/null` redirection to prevent the error output.
Also the script shouldn't be running in first place since you've deactivated WiFi, I'll see to fix it in the next versions.

Krocko, 2018-10-10:
Have updated to v2.05.40. Now i get this. 
`Oct 10 17:28:51 raspberrypi deCONZ-WIFI.sh[407]: /usr/bin/deCONZ-WIFI.sh: Zeile 874: /: Ist ein Verzeichnis
Oct 10 17:29:21 raspberrypi deCONZ-WIFI.sh[407]: /usr/bin/deCONZ-WIFI.sh: Zeile 874: /: Ist ein Verzeichnis
Oct 10 17:29:52 raspberrypi deCONZ-WIFI.sh[407]: /usr/bin/deCONZ-WIFI.sh: Zeile 874: /: Ist ein Verzeichnis
`