dresden-elektronik/deconz-rest-plugin #509: repeating schedule works only for groups?

ErnstEeldert, 2018-03-27:
I am having trouble setting up schedules where the command is a light or a sensor. If it's a group, it behaves as expected.

I have setup a repeating schedule like this:

`{
  "activation": "start",
  "autodelete": false,
  "command": {
    "address": "/api/username/sensors/33/state",
    "body": {
      "flag": true
    },
    "method": "PUT"
  },
  "created": "2018-03-27T09:46:44",
  "description": null,
  "etag": "b7e695ea7e11e3e328922116bb19f2a6",
  "name": "New Schedule",
  "status": "enabled",
  "time": "W127/T11:07:00"
}`

Nothing happens. I am aware of the UTC vs localtime difference.

If the command is point to a group, it works as expected. Is there a way to get some deconz logging on this topic?

manup, 2018-03-27:
Hmm haven't tried it myself but the code suggests sensors should work.

[image]

>If the command is point to a group, it works as expected. Is there a way to get some deconz logging on this topic?

The debugg logs can be enabled by starting deCONZ like this:

    $ deCONZ --http-port=80 --dbg-info=1 --dbg-http=1

ErnstEeldert, 2018-03-27:
Hmm. Tried again with a sensor, and it indeed seems to work. This is what the logs tell me:

`12:43:00:469 schedule 2: New Schedule trigger
12:43:00:469 schedule 2 body: {"flag":true}
12:43:00:486 discard sensor state push for state/flag (already pushed)
12:43:00:487 ZCL attribute report 0x90FD9FFFFE32B85D for cluster 0x0008, ep 0x01
12:43:00:488 0x90FD9FFFFE32B85D level 0 --> 102
12:43:00:493 discard sensor state push for state/lastupdated (already pushed)
12:43:00:493 rule event: /sensors/33 state/lastupdated num (0 -> 0)
12:43:00:494 trigger rule 62 - HomeOffice.Lights.TurnOn.Night`

Where I got confused was that I was watching the websocket stream, and the sensor state change didn't show up. Even I do something like `$ PUT /sensors/33/state '{ "flag": false }'
[{"success":{"/sensors/33/state/flag":false}}]`, nothing is shown on the websocket stream. The state gets persisted alright though, and rules are executing as well. Expected behavior?

However, when I change the schedule command to operate a light, nothing happens. Here's the schedule:
``
Here's the log:
`13:01:58:759 discard sensor config push for config/temperature (already pushed)
13:01:58:762 discard sensor state push for state/presence (already pushed)
13:02:00:374 binding for cluster 0x0006 of 0x000B57FFFEC0789C exists (verified by reporting)
13:02:00:375 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x000B57FFFEC0789C (seems to be active)
13:02:00:376 binding for cluster 0x0008 of 0x000B57FFFEC0789C exists (verified by reporting)
13:02:00:376 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x000B57FFFEC0789C (seems to be active)
13:02:00:378 Poll APS request to 0x000B57FFFEC0789C cluster: 0x0006 dropped, values are fresh enough
13:02:05:374 binding for cluster 0x0006 of 0x000B57FFFEBE26FF exists (verified by reporting)
13:02:05:375 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x000B57FFFEBE26FF (seems to be active)
`

{
  "activation": "start",
  "autodelete": false,
  "command": {
    "address": "/api/username/lights/4/state",
    "body": {
      "bri": 255,
      "on": true
    },
    "method": "PUT"
  },
  "created": "2018-03-27T09:46:44",
  "description": null,
  "etag": "56a1885b30856a9923cf3cba59f51eaf",
  "name": "New Schedule",
  "status": "enabled",
  "time": "W127/T12:02:00"
}

closed 2019-03-30 by stale[bot]

mc7547, 2021-06-18:
Good morning
could I have an example of a schedule that turns on light 4 at 18:00:00?
thank you
roby