dresden-elektronik/deconz-rest-plugin #511: Loosing Connection to CONBEE while using another  Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller

basty2, 2018-03-28:
I'm using Conbee and it works. But if I connect a CP2102 deconz program is loosing the connection to the usb stick. 
`usb-FTDI_FT230X_Basic_UART_DO00KCJP-if00-port0 -> ../../ttyUSB0
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB1`
The CP2102 works fine. If I disconnect the cp2102, deconz works again. Is it possible to use both sticks?

manup, 2018-03-28:
Hm strange once deCONZ has a connection to ConBee it should keep it open, so normally if you connect the other dongle after connection is established it should still work.

Can you please provide the output of `$ lsusb`?

basty2, 2018-03-28:
only conbee stick: 
`~$ lsusb
Bus 002 Device 003: ID 174c:5106 ASMedia Technology Inc. ASM1051 SATA 3Gb/s bridge
Bus 002 Device 005: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 002 Device 004: ID 05e3:0743 Genesys Logic, Inc. SDXC and microSDXC CardReader
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 107: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 114: ID 0bda:0153 Realtek Semiconductor Corp. Mass Storage Device
Bus 001 Device 003: ID 8087:0a2a Intel Corp.
Bus 001 Device 078: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
with cp2102 

`~$ lsusb
Bus 002 Device 003: ID 174c:5106 ASMedia Technology Inc. ASM1051 SATA 3Gb/s bridge
Bus 002 Device 005: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 002 Device 004: ID 05e3:0743 Genesys Logic, Inc. SDXC and microSDXC CardReader
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 107: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 118: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 117: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 8087:0a2a Intel Corp.
Bus 001 Device 078: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

manup, 2018-03-28:
Sorry I don't know what is causing the problem I thought both devices where FTDI based with IDs 0403:6015 which could confuse deCONZ but this doesn't seem to be the case here.

basty2, 2018-03-28:
Maybe this helps. at 08:37 I connected the cp2102 and got errors.

[image]

manup, 2018-03-28:
It looks that the connection drops out, maybe there is some udev rules script which triggers then the other dongle is attached.

basty2, 2018-03-28:
thank you. it seems a problem with fhem, steering the homematic stick and conbee. I have to take a look tomorrow.

basty2, 2018-03-29:
Hi. I don't really know what is the reason. after flashing the firmware on both sticks it seems to be working again...

closed 2018-03-29 by basty2