dresden-elektronik/deconz-rest-plugin #512: IKEA Trådfri motion sensor battery drain

sveip, 2018-03-29:
After some unscientific measurements it seems that the motion sensor is loosing battery faster when connected to Deconz compared to the IKEA hub.

Might there be some polling going on that can cause this and preventing it to sleep?

manup, 2018-03-29:
Nope, the sensor works on its own and sends out just the motion sensed commands and battery level reports. Do you have the latest firmware of the sensor?

sveip, 2018-03-29:
Thanks for the fast replay. Ok, was probably difference in the batteries (poorer batteries when on Deconz.)

Is there a way to update the firmware from deconz, or do I have to move it back to the IKEA hub?

closed 2018-03-29 by sveip

sveip, 2018-04-13:
@manup, is the update feature just present in the Qt UI, or also web? Can't seem to find it in webui, and I run headless.

manup, 2018-04-13:
For battery powered end-devices, currently it can only be activated via gui.