dresden-elektronik/deconz-rest-plugin #513: High cpu usage? ZDP skip fetch

jonatanolofsson, 2018-03-29:
On my RPi, the deconz process is taking up about 30-50 % of CPU resources. I do have quite a large setup (30 lights, 50 sensors) so I'm not sure if it's to be expected, but running with debug output, one message stands out in particular:
```18:23:21:982 ZDP skip fetch, unconfirmed zdp request count 2 [2]```
I could not find anything in the (open) source that output this, but it seems to be outputting at about 10-20 Hz when it occurs, though not all the time. Is this a retry of some sorts that never succeeds, or is this message expected?

manup, 2018-03-30:
The ZDP message is expected or at least not uncommon, it comes from the ZDP timer tick which runs every 80 ms. Since there are already open requests (unconfirmed on ZigBee level) the tick decides to skip doing more requests for this specific device. The ZDP tick is using a forward walking iterator so that it only processes on device per tick to scale also in large networks.

30 % CPU usage might be related to latest GUI updates which uses a more expensive drawing scheme. When you minimize the deCONZ window the CPU load should go down (same if you running deCONZ as headless).

ErnstEeldert, 2018-03-30:
I've noticed this too. Running in headless mode on a rpi3 with 52 lights and 30 sensors connected and 37 rules defined.

`load average: 0.26, 0.29, 0.28`

CPU usage of deconz never seem to go below 10%, and is usually hovering around 30%.

jonatanolofsson, 2018-03-30:
For the record, I'm running headless

jonatanolofsson, 2018-04-25:
IIRC, I turned out to be the culprit (my application forgor to await the sleep routine).

closed 2018-04-25 by jonatanolofsson