dresden-elektronik/deconz-rest-plugin #514: Unique id  for groups

Kane610, 2018-03-29:
Would it be possible to generate a unique id for groups (all virtual devices) on creation? This would help with keeping in line with other devices and would help improve the integration with home assistant.

marthoc, 2018-04-05:
Hi @manup - it would also be helpful for the Home Assistant integration if the Daylight sensor had a uniqueid generated for it. All other sensors should have a uniqueid (I think all hardware-based devices would and CLIP sensors should since the uniqueid field is required when creating a sensor over the api). But since Daylight is an automatically created sensor, there is no opportunity to specify a uniqueid and that field can't be added via a REST call. I do realize that the Hue API (on which the Daylight sensor seems to be based) does not generate a uniqueid for that sensor either (an issue I have raised on the Hue developer forums too).

manup, 2018-04-05:
Sure I'm fine with giving it a uniqueid, any suggestions? — something mac address like *00:00:00:00:00:00:00::01*?

marthoc, 2018-04-05:
Yes a MAC address would be fine, that would make it similar to other devices. Since it's unique to the bridge (I would think there is only one of these since it's created automatically) it's uniqueid could also be based on the bridge's UUID or something similar?

manup, 2018-04-05:
Yes it could be derived from the bridge mac address: *00:21:2e:ff:ff:xx:yy:zz-01*

marthoc, 2018-04-05:
That sounds like it would work well! Thanks.

Kane610, 2018-04-07:
@manup I don't get any uniqueid reported when requesting /api/**key**

marthoc, 2018-04-07:
@kane610 I think you may have to delete and recreate the group?

Kane610, 2018-04-07:
@marthoc did they add uniqueid for groups recently? I moved everything over just over a month ago.

Kane610, 2018-04-07:
@marthoc then disregard previous comment since I haven't upgraded, didn't know it was fixed.

Kane610, 2018-05-07:
Hi!

As far as I can tell I still don't get unique ids for groups. Even though I've added new devices and created new groups.

/R

Kane610, 2018-07-31:
I have tried and played around and I still don't get unique ids for groups. Am I doing something wrong?

closed 2018-12-05 by stale[bot]

helto4real, 2019-05-19:
I tried to add new group to deconz and I did not get id of the group. If so Home Assistant would pick it up as new device. If it not show as device I cant use it in areas and have to manage individual lights in my automations. Is it OK if someone check this one out again? @Kane610 Is this not correct?

Kane610, 2019-05-19:
@manup this issue is still relevant I believe

manup, 2019-05-21:
Just checked the code, currently a `uniqueid` attribute in form of  `AA:BB:CC:DD` or `AA:BB:CC:DD-XX` can only be specified when a group is created.

Otherwise the normal `id` attribute should be used, since it already *is* a uniqueid within the Zigbee space.

helto4real, 2019-05-21:
Yea strange that I don't seem to get unique id when I create new group. You think it can have to do with that there are older groups there too?

justme-1968, 2020-04-17:
please make sure uniqueid is unique or empty. never have more than one device with the same id.