dresden-elektronik/deconz-rest-plugin #515: Xiaomi Temp/Humidity Sensor

jaal2001, 2018-03-31:
Hi,

I got the first version of the Xiaomi temperature and humidity sensor today, but I am not able to add it via the webinterface.
As far as I understood the sensor is supported, do I need the Xiaomi bridge to add the sensor intially or should I be able to add it via the webinterface directly? I tried to reset the sensor by pressing the button >5 sec, but it still does not appear in the webinterface.

Can anybody give me a hint on this?
Thanks
J

tleegaard, 2018-03-31:
In PWA go to Sensors -> Add sensor -> Other. Hold down the reset button for ~4 seconds, until the Xiaomi sensor blinks. Wait a couple seconds and the sensor should blink twice if succesfully paired. Afterwards it will show in both the webinterface and API.

jaal2001, 2018-04-02:
Well I did that, but the sensor does not show in the webinterface. I can see 4 sensors (I guess I tried it 4 times) in the config (REST export)but not in the webinterface. I do only see lights and switched on the webpage. I'm using 2.04.99, do I need to upgrade to a newer version (Beta)?

tleegaard, 2018-04-02:
Not sure when support was added. I'm on 2.05.17 myself (and just a regular user ;-)).

jaal2001, 2018-04-08:
Okay, it seems that support has been addes after 2.04.99, so I switched to the latest beta and now it works. Thanks!

closed 2018-04-08 by jaal2001