dresden-elektronik/deconz-rest-plugin #516: Can't connect IKEA bulb

pfassberg, 2018-04-02:
I've been playing around a bit with a few IKEA bulbs using 2.04.99.  It almost worked but I could'nt change both colour and brightness in the same REST command.  The brightness was never changed.  If I did it in two separate commands it worked.

I'm using a RPi and a ConBee stick.

I decided to try bleeding edge: deCONZ 2.05.17 and fw 0x261e0500.  The IKEA RGB bulb already had the latest firmware: 1.3.002.

Upgrading went well but it didn't work at all,  It didn't find any lights nor any sensors.  The lights were totally gone but the sensors were only "disconnected". I could see the sensors in the web app and the REST api. I decided to do a factory reset, remove the databases, reinstall deCONZ and reconnect the bulbs. Unfortunately I can't reconnect any bulbs.  I downgraded to 2.05.16, still the same.  I downgraded to 2.04.99 that worked before, still no change.

Is there any known bugs in 0x261e0500 regarding IKEA bulbs or what am I doing wrong?  From what I found on this forum that firmware is the best for IKEA bulbs.

I decided to see if I could notice anything in the X11 app, but the only strange here is that I see 3 devices despite doing a factory reset, and the NWK is 0x000 for all of them in 2.05.16/17 but different numbers in 2.04.99.  These 3 devices are probably my 3 IKEA sensors (motion control, remote control and a dimmer).  

2.04.99:
[image]

2.05.17:
[image]

I'm stuck.

Probably I need to downgrade the firmware but which should I use?

// Peter

manup, 2018-04-02:
>I've been playing around a bit with a few IKEA bulbs using 2.04.99. It almost worked but I could'nt change both colour and brightness in the same REST command. The brightness was never changed. If I did it in two separate commands it worked.

If i remember correctly that's a bug in the IKEA bulb firmware we need to wait for new bulb firmware here, until then only separated commands will do.

2.05.17 and 0x261e0500 are fine and recommend to use. But somehow your settings got scrambled. Note after a firmware update it might take a while till all devices reappear (but less than 2 minutes). 

To reconnect the bulbs you need to reset them with the IKEA reset sequence *while* light search is running in Phoscon App.

https://www.youtube.com/watch?v=mJm9YpPrGzk

What are these 3 nodes? The mac address seems unfamiliar..

pfassberg, 2018-04-02:
The 3 nodes are actually some Samsung SmartThings sensors. Did not show up in the WebApp nor the REST api so it was i bit of a surprise that they show up in the X11 GUI.

I have reconnected the bulbs numeruos times when using the old s/w and it worked (almost) every time.

Is the PWA doing anything better then the old Web app?  I tried PWA now to no avail:
[image]
However, the light is showing up in the X11 GUI.

I decided to do another try so I once again did a factory reset and deleted all the databases,

This time I got a new PAN id and still nothing worked. I noticed that the Type was now "Router" so I changed it to "Coordinator" and did another search and voila! It worked!!

Now I have a working setup again.

pfassberg, 2018-04-02:
And a note: The comment about IKEA reset in PWA is wrong:
[image]
The IKEA lights should be toggled 6 times to be reset.

sveip, 2018-04-21:
I'm also struggling to include Ikea trådfri bulbs. They are all on latest Ikea firmware versions.

auspex-nl, 2018-04-22:
I had similar pairing problems with the Ikea bulbs, no problems with the remotes.
What worked for me (2 bulbs connected now) was the following: 

- set the light search period to 10 minutes in the settings (anything higher than 1 minute should do)
- get the light bulb in reset mode (do the reset sequence, turn on/off 6 times, this I found to be tricky, see comments below)
- click on Add new light and let it search
- now turn the bulb off and on once more
- wait (can take a few minutes (!!))

Getting the light bulbs in reset mode was quite tricky actually, i'm not sure how fast to turn mains on/off etc. In the old interface (http://hass:8040/edit_system.html -> use your own url of course, might be localhost:80) there's an option to reset devices using Touchlink, you'll need to bring the bulb within 50cm from the RaspBee, scan for devices on this page and then press the red Reset button. For me, this puts the bulbs in factory reset mode and is much easier than the Ikea 6 times on/off sequence.

pjcarly, 2018-09-29:
@vuurbeving That worked perfectly, just connected 2 new Tradfri color lights just like you described. Any other way didnt work. Thanks!

BoneheadFraggle, 2018-10-13:
> I had similar pairing problems with the Ikea bulbs, no problems with the remotes.
> What worked for me (2 bulbs connected now) was the following:
>
>     * set the light search period to 10 minutes in [...]

Thanks a lot for this. I actually had to adjust the process to add my Trådfri 1000 lm (https://www.ikea.com/se/sv/catalog/products/60338452/). I noticed after a lot of testing that if I added them to a Trådfri remote first, then reset the lamp - then it was fast and easy to add to Conbee. If left our the remote pairing, it was a lottery to get it in to deConz.

1. Reset the Trådfri remote (4 x reset button). The remote flases 4 times.
2. Push the on/off button on the Trådfri remote just make sure it is awake
3. Pair the remote with the lamp: Turn on the lamp, Hold the remote very closely and hold the reset button on the remote until the lamp stops flashing. No hurry, let it complete.
4. Dim the lamp to minimum (this makes it easier to make sure the lamp is successfully reset in the next step)
5. Turn the lamp on/off 6 times with the manual switch (or pull the plug) and leave it on the 6th time. It should blink and then go to maximum brigtness. The lamp is now reset to default.
6. Leave the lamp on, put it close to your conbee.
7. Add new light
8. Wait a couple of seconds, then turn the lamp off and then on with the manual switch
9. Phoscon will now find your lamp within seconds.

For me it worked 4 times in a row with different, brand new lamps.

anoxi, 2018-12-31:
@BoneheadFraggle Thanks, this finally works for me with deCONZ and the phoscon web app. Just a note on step 5: i had to turn the lamp on/off very quickly

closed 2019-05-07 by stale[bot]