dresden-elektronik/deconz-rest-plugin #518: Network issues in 0x261f0500

ebaauw, 2018-04-03:
I've been running v2.05.17 with firmware 0x261f0500 for a week or so.  Last night the old network issues re-appeared:
- Red nodes in the deCONZ GUI; pressing `0` doesn't always bring them back.  Especially the Hue color lights (Hue bloom, old Hue lightstrip) seem lost, even after power cycle;
- Hue motion sensors, Hue dimmer switches, I remotes blink red (presumably because they don't get an ACK from the gateway); the Hue dimmer switches continue to work standalone;
- I reset the RaspBee using `sudo GCFFlasher_internal -r`, but no joy.  It takes a ridiculous long time (> 30 mins) to re-discover the entire network, some nodes show red again, some remain lost;
- deCONZ itself remains fully functional: REST API is responsive, rules and schedules are fired, etc.

I downgraded to 0x261e0500 and, like magic, the issues disappeared.

manup, 2018-04-03:
Hmm strange, not sure if it is the firmware, might also be a queue issue in deCONZ itself. After the `sudo GCFFlasher_internal -r` and before downgrade to 0x261e0500 did you also restart deCONZ?

ebaauw, 2018-04-03:
Yes, I shutdown deCONZ before resetting the RaspBee, then restarted deCONZ.

Anyways, now running v2.05.17 with 0x261e0500.  If the issue re-appears we'll know it's not the firmware.  0x261f0500 did run for a week (or so) without any issues.

manup, 2018-04-03:
I have a few larger installations on 0x261f0500 running for approx a week now, I'll keep an eye on them. It's very strange that after RaspBee reboot and deCONZ restart the problems didn't resolve.

ebaauw, 2018-04-03:
Strange, maybe, but not for the first time: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/33#issuecomment-330633123

ebaauw, 2018-04-22:
I've been running firmware 0x261e0500, now with v2.05.21.  I haven't seen any issues until yesterday evening.  No red nodes, but all my Hue motion sensors were blinking red.  The Hue dimmer switches continue to work standalone.  I didn't have the time to do a full analysis, but again stopping deCONZ, resetting the RaspBee and starting deCONZ didn't solve it.  I had to power off most of my lights, stop deCONZ, reset RaspBee, start deCONZ, gently power on lights.

Not sure if this is related, but after this, the GUI wasn't showing the full endpoints and cluster for my ubisys dimmer.  On some endpoints, the device IDs was 0xffff and some of the clusters where missing.  Waited overnight, tried opening the network (from the web app), power cycling the dimmer, removing the node from the GUI, pressing 0, but nothing helped.  It was corrected only when I did _Edit_|_Reset selected nodes_ from the GUI.

alexvandervegt, 2018-05-25:
I also have a lot xiaomi wireless switches who spontaneous loss the network. I need to delete them and re-add them to work again. But then a few hours later they can lose the network again.

manup, 2018-05-25:
To which parent nodes are the sensors connected? With other end devices we found that IKEA lights follow strict ZigBee 3.0 "rules" and remove end devices if they don't poll in a certain time, after that end devices must re-join the known network in a specific manner &mdash; there seems to be some issues in some end devices in regard to this.

Currently the only way to prevent this is to force the sensors to connect to more tolerant devices like Philips hue or the gateway directly.

auszs, 2018-09-01:
Can reproduce the problem with deCONZ 2.05.20, ConBee and the above mentioned firmware 0x261f0500 running on Ubuntu 18.04.
- Nodes (in my case various battery powered Xiaomi sensors) work fine after pairing.
- The network is recovered properly after restarting the deCONZ application - the particular device appears in the GUI as soon as the first packet is received.
- The connection seems lost (log showing "unknown node") after closing deCONZ, disconnecting the USB dongle, and starting deCONZ again.

Sniffing the ZigBee traffic - packets sent by the device look fine. The network connections shown in the deCONZ UI look identical after the application restart.

ebaauw, 2018-09-07:
Should be solved 0x26230500, see #771.

closed 2018-09-07 by ebaauw