dresden-elektronik/deconz-rest-plugin #519: xiaomi aqara motion sensor intermittent behavior

ErnstEeldert, 2018-04-03:
I am having troubles getting Xiaomi Aqara motion sensors to behave properly. The behavior that I am seeing is that motion events are not being sent or received intermittently. As a result, state/presence is set to false incorrectly by deconz (based on config.duration).

I've swapped out the sensors to check if the sensor itself was faulty, but the erratic behavior is consistent.

Here's an example of state.presence value over time for a sensor with config.duration=90. I've set it to a value significantly larger than 60 to avoid incorrect state/presence changes, as the heartbeat interval of the sensor is ~62 seconds.

`$ while sleep 5; do GET /sensors/20 | jsawk "return new Date() + ' ' + JSON.stringify(this.state)"; done
Tue Apr 03 2018 12:49:15 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:20 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:25 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:30 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:36 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:41 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:46 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:51 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:49:56 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:49:00","presence":true}
Tue Apr 03 2018 12:50:01 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:07 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:12 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:17 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:22 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:27 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:32 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:37 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:43 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:48 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:53 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:50:58 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:03 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:08 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:13 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:18 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:24 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:29 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:50:02","presence":true}
Tue Apr 03 2018 12:51:34 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:51:39 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:51:44 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:51:49 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:51:54 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:51:59 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:52:05 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:51:32","presence":false}
Tue Apr 03 2018 12:52:10 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:15 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:20 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:25 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:30 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:35 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:41 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:46 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:51 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:52:56 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:01 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:06 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:12 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:17 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:23 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:28 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:33 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:38 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:52:10","presence":true}
Tue Apr 03 2018 12:53:43 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:53:48 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:53:53 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:53:59 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:54:04 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:54:09 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:53:41","presence":false}
Tue Apr 03 2018 12:54:14 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:54:13","presence":true}
Tue Apr 03 2018 12:54:19 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:54:13","presence":true}
Tue Apr 03 2018 12:54:24 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:54:13","presence":true}
Tue Apr 03 2018 12:54:29 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:54:13","presence":true}
Tue Apr 03 2018 12:54:35 GMT+0200 (CEST) {"lastupdated":"2018-04-03T10:54:13","presence":true}
[... 110 lines cut ...]
    {
      "address": "/sensors/20/state/presence",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2018-03-26T09:49:19",
  "etag": "80ab2bd9e4960842ddefb3a746665800",
  "lasttriggered": "2018-04-03T11:51:39",
  "name": "Bathroom.Door.ClosedNoPresence",
  "owner": "xxx",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 92
}
`

I don't see how the rules could affect the behavior though, as I just don't see the state/presence being updated correctly in the first place.

Vendor dresden elektronik
Product RaspBee
Version 2.05.15 / 3/18/2018
Firmware 261E0500

Ideas?

manup, 2018-04-03:
Just tried it with mine, confirmed via sniffer that the sensor *most of the times* updates every 60 seconds. Looking closer in the sniffer logs it seems that sometimes the sensor doesn't reach out even with impressive dancing actions involved.

I've seen gaps up to 5 minutes (only when permanent motion should be detected) going from non presence to presence initially always works.

So I think to be on a safe side you may rise config.duration to at least 120 or even up to 300.

ErnstEeldert, 2018-04-03:
Thanks for confirming the sensor being not as reliable as we would like it to be. I will increase config.duration and see if I can get it to leave the lights turn on.

Any recommendations on a reliable motion sensor also appreciated. Still having issues with the Philips Hue motion sensor (#386), and the IKEA Tradfri one doesn't seem to be 'sensitive' enough from a hardware perspective, ie. I need to be quite close for it to detect motion.

ErnstEeldert, 2018-04-10:
Just thinking, couldn't it be a generic motion sensor issue in combination with deconz? The reason I'm saying that, is because I cycled through 5 different Xiaomi sensors, and they all seem to show the same behavior. They have been shipped from the PRoC separately from different vendors, so it's unlikely they're from the same production badge.

On top of that, I have the Philips motion sensor, which also shows the intermittent behavior. Thing with the Philips is, it actually shows you the red light of death if it's failing to communicate with the mesh.

Unfortunately, I don't have a Hue or Xiaomi bridge to confirm all sensors are good otherwise.

I can't recall to have this behavior from when I started using the sensors, so my guess is some kind of firmware/deconz upgrade introduced this behavior.

Thing is, it is becoming a showstopper quickly in this house, as people get annoyed by the lights not responding, so resorting to the always reliant wall switch 🙈

ErnstEeldert, 2018-04-11:
One more thing I wanted to add here, I also have a bunch of Xiaomi door sensors (v1 & v2), and they work flawlessly.

manup, 2018-04-12:
So the issue remains with a higher `config.duration` value? My Xiaomi motion sensor works very reliable (kitchen entry) `config.duration` is set to 180, the rules are set via Phoscon App motion sensor control.

sweebee, 2018-04-16:
I have mine at default, but why does it reset at 60s when there is still motion? de reset timer should reset when there motion within the 60s. It should be possible because this also works with the default aqara gateway and in Homey.

ErnstEeldert, 2018-04-17:
I have the config.duration set to 90 or 120 seconds. Doesn't matter (60s is not good though, as it interferes with the Xiaomi heartbeat of 60s). Often it just doesn't trigger to begin with. I walk into the bathroom, in front of the sensor, no lights turned on. I stay in the bathroom, and after a while (not a fixed time) they lights turn on. I checked the event log on the websocket, and the presence is just not there.

The rules I have setup manually, as I use a door sensor in combination with the presence sensor have the lights go on as soon as you open the door.

closed 2019-03-30 by stale[bot]

sweebee, 2019-03-30:
I use this simpel “hack”. It works great. 

After connecting the pins, it Will trigger motion more often. 

https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

MattL0, 2019-09-15:
how's the battery life with this hack? thanks

@sweebee