dresden-elektronik/deconz-rest-plugin #522: Can't find lights (or Conbee?) after update

JohnEriksson00, 2018-04-04:
I updated the software to the latest version, but not no lights can be found (even after reseting the light and opening the network). When I access the webGUI I see that my channel is 0 and my network ID is 0x0000, which leads me to belive that the connection to the Conbee stick is at fault?

I am running a Conbee stick on a Raspberry Pi running Hassbian (Raspbian with HASS). No other USB-devices, and I tried moving the stick and rebooting.

manup, 2018-04-04:
Did it work before the update or is this a fresh installation?

JohnEriksson00, 2018-04-04:
It did work before the update.

JohnEriksson00, 2018-04-04:
I wiped the SD-card, installed a fresh Hassbian.
Then I did; 
 `sudo apt install qt5-default libqt5sql5 libqt5websockets5-dev libqt5serialport5-dev wiringpi`
Before installing both deconz-2.05.16-qt5.deb and deconz-dev-2.05.16.deb. 

It still says all 0 on the web GUI and I can't get my Trådfri lamp to connect..

manup, 2018-04-04:
All 0 means that either the network is not operating or the connection to the dongle isn't working. Do you have the deCONZ gui on Hassbian to check the deCONZ network settings?

JohnEriksson00, 2018-04-05:
I do not have a GUI installed, Hassbian is only commandline. Is there anything else I can do other than install a dist with GUI? I tried with a different zigbee component, it still won't connect.

manup, 2018-04-05:
Is the current firmware version shown in the Phoscon App (meaning connection to the dongle is working)?

If so, you may try to reset the gateway in *Phoscon App > Menu  > Gateway Settings*. Devices must be rejoined to the network then though.

I'm currently working on much more stable update and auto backup settings process so that such things won't happen anymore, hope to get it done in the next versions.

JohnEriksson00, 2018-04-06:
[image]
[image]
 These are the only two views available to me in the Phoscon App, not sure where to find the menu, if it is supposed to be visible?

manup, 2018-04-06:
Ah I see, jep currently you can't pass the step above, we will add something for that soon.

As a workaround you can create a group in the old webapp, then the Phoscon App will skip the above screen.

To open the old webapp use:

http://192.168.1.8/login.html

JohnEriksson00, 2018-04-06:
Even after a reset the firmware is 0000, se attached screenshot. Also, pressing update results in 

> Update failed. Please try again later.

[image]

manup, 2018-04-06:
Hmm hard to tell, I'll put up a Hassbian on a spare RPi and see whats going on and if there are ConBee specific differences to Raspbian.

JohnEriksson00, 2018-04-06:
So I got it working by flashing [Hass.io](https://www.home-assistant.io/hassio/installation/) and adding the deconz plugin. This is a different dist, so the issue with Hassbian could still persist. I will continue to use Hass.io as it seems like it is good enough for my needs at the moment..

uli6, 2018-05-17:
I'm having the same problem, but with hassio and RaspBee.

Paul-dH, 2018-10-07:
I have the same exact issue, tried to roll back to Hassio 79.0 and had no luck. Even 78.x didn't work anymore. I did get Phoscon app working again and my gateway was found. Everything looks ok and the logfiles indicate that all went well. But only all the lights stay on status "Unavailable", in Hass and in Deconz. All the switches and Dimmers are available and found, in the JSON reponse of the API all lights have: reachable: false in the state block.

I also did a test with Deconz on my Windows laptop and the ConBee, in that setup everything works and all switches and lights are available. I don't get it.

Is there something I can do to help?

geekuality, 2019-01-12:
I'm having this problem too with a fresh hass.io installation (RPi 3B+ and RaspBee). My objective is only to use ZigBee for reading sensors, so even having to visit Ikea to get a light bulb to complete the installation was mind-boggling and now the system does not even detect the bulb.

manup, 2019-01-12:
In future versions the Phoscon App will have a way to skip the commissioning page.

As workaround you can manually open the http://dresden-elektronik.de/pwa/index.html page (you need to be logged in). Once the first group is created this main page will be opened in further logins.

Gustra, 2019-04-13:
I am also having the same problem on hass.io (version 159), and using the old login page does not work. I did manage to get the login form, but after submission the Phoscon app redirected to /index.html (not through the ingress). The "old" index page redirects to the new login page. I have tried accessing the edit_group.html page also, but and it redirects to the new site before I have a chance to create a new group. Is there any other way to create a group, if that it the key to get it working? Perhaps using a REST API call?

manup, 2019-04-13:
To get to the old WebApp login page you can use `/login.html`
the new Phoscon App login page is located at `/pwa/login.html`

After the first login you can skip commissioning and create a group on the next screen.

Gustra, 2019-04-13:
I have a clean install of hass.io and deConz in RPi 3+. I try to login using the default delight/delight information using the http://192.168.0.92:8123/api/hassio_ingress/woWb4LkxCq3tBiAqaSLweLaLuniu6uOPdoar_v6TKRA/login.html page. After login the browser is redirected to http://192.168.0.92:8123//index.html. Also tried logging in using the user for the hass.io installation. Either way, accessing .../pwa/index.html or .../index.html redirects to the .../pwa/login.html page. I don't get the opportunity to create any user/password for Phoscon at all after opening the WEB UI on the deConz plugin.

closed 2019-08-18 by stale[bot]

ronnyandre, 2020-07-01:
@Gustra Same issue here. Is the old GUI not available?