dresden-elektronik/deconz-rest-plugin #523: Raspbee not working

sweebee, 2018-04-04:
I got a raspbee module. The web app says:

```
Your gateway is on channel 15 and has the network id 0x7347.
```

If I try to change the channel to 20, nothing happens. 

When scanning for devices nothing is found and 10 seconds later the message "updating" will show again (like its updating the firmware).

I tried to connect a hue bulb, trafri bulb and a Xiaomi smart switch but nothing works.

manup, 2018-04-04:
Which deCONZ version are you running?

manup, 2018-04-04:
I figure the RaspBee has old firmware which needs to be updated normally a update button should be shown.

A few things to check:

1) Is your serial port configured as described here?
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements 

2) Check RaspBee firmware version in Phoscon App

`http://<raspberry-ip>/pwa`

Open *Hamburger Menu > Settings > Gateway*, there the current firmware version is displayed.

sweebee, 2018-04-04:
I checked the raspi-config, but there is no ```Interfacting Options > Serial``` but I found: ```Advanced Options > Serial``` where I can set "Would you like a login shell accessible over serial" to No and than It says ```Serial is now disabled``` and goes back to the main screen, so I don't see "Would you like the serial port hardware to be enabled?".

manup, 2018-04-04:
Just to be sure, you have Raspbian Jessie or Stretch? Wheezy is no longer supported.

You can see Raspbian version here:

`$ cat /etc/os-release`

sweebee, 2018-04-04:
```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
```

manup, 2018-04-04:
That's strange, I wonder why is the raspi-config so different

sweebee, 2018-04-04:
I'' try to upgrade everything and if that doesn't work I try another raspberry with a clean image.

sweebee, 2018-04-04:
After upgrading raspi-config dit match, but no luck. I'm trying to install the official image now.

ebaauw, 2018-04-04:
Did the bulb react to the reset?  If it's on the latest Hue firmware, you need to power-cycle it just before touchlinking, or it won't reset.

sweebee, 2018-04-04:
Solved, when i powered off the hue bridge everthing worked fine.

closed 2018-04-04 by sweebee