dresden-elektronik/deconz-rest-plugin #525: WF323 support on deconz

vamanea, 2018-04-05:
On some sites one can buy a Zigbee LED strip controller called WF323(I won't link here but google will provide).

I've been doing some sniffing and found out they are probably just doing the most basic ZStack implementation from TI with a CC2530 chip. After a bit of head banging and with a help of a radio monitor I managed to get it to join my raspbee network. It seems to only join the network if security is completely disabled: Security Mode : No Security
It also doesn't seem to work as a RGB light just a normal light(has a level) but it does show up in the webpage.

I also have a pcap file from Wireshark with the communication if anybody is interested. To get it in some proper shape it probably needs some new firmware with security enabled.

manup, 2018-04-05:
>I also have a pcap file from Wireshark with the communication if anybody is interested. To get it in some proper shape it probably needs some new firmware with security enabled.

Absolutely since disabling the security isn't really a option, maybe it supports both modes and also joins a HA default TC link key secured network like most devices do.

vamanea, 2018-04-06:
[wf323.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1884998/wf323.zip)
Here are my pcap files created by Wireshark:
* wf323_nosec and wf323_nosec_comms are captured when I disabled security for my whole network in deConz. The WF323 seems to join the network, I can see the node in the GUI, I can control the light in the web interface.
* wf323_sec_enable is a capture of the trafic when Security is Enabled(TC Link Key) It takes a while until I convince the device to try to join the network but on the first encrypted payload from raspbee the comunication just seems no to work anymore.

The WF323 MAC is 00:12:4b:00:12:da:4a:e1 - doesn't seem they did too many customizations to the TI stack.

troylynch, 2018-05-05:
How do you disable security in deconz? I would like to play around with this.

vamanea, 2018-05-06:
[image]

In network settings just set **Security Mode** to **No security** though as @manup has said it's not a good thing

troylynch, 2018-05-07:
Thanks. Found it. I was running headless is why. I'm still not able to get the wf323 to be seen by the raspbee. Is there a trick I'm missing?

vamanea, 2018-05-07:
I used a ZigBee monitor to checkout the traffic. At first the WF323 was only trying on channel 25. I just changed my Raspbee to channel 25, no security and open network. IIRC you just power it on and press buttons 1 & 3 for some time and it will join the open network.

closed 2019-03-30 by stale[bot]