dresden-elektronik/deconz-rest-plugin #526: Heiman HS1RC Smart Remote Control

oleg77s, 2018-04-06:
Heiman HS1RC Smart Remote Control and Heiman HS1EB Smart Emergency Button support?
@ebaauw: I'd expect these to be IAS Zone devices, just like most of the other Heiman sensors. If you can provide me screenshots, adding support shouldn't be too difficult. Probably best to expose them as ZHASwitch and issue appropriate buttonevents for press/release of each/the button.

[image]
[image]
[image]
[image]
[image]

closed 2019-03-30 by stale[bot]

accelle17, 2019-07-22:
@oleg77 Were you able to integrate the HS1RC on deconz, particularly on homeassistant?

olix74, 2020-02-22:
I would also love to see support for the Heiman Smart Remote Control!
Thank you,
Oliver