dresden-elektronik/deconz-rest-plugin #529: Condition don‘t work anymore

Krocko, 2018-04-10:
@manup: If i set the following condition in a rule, the rule don‘t work anymore. Have you a solution for this? I have deCONZ 2.05.20 installed. I mean, that this condition worked in the past. 

{
        "address": "/lights/5/state/on",
        "operator": "eq",
        "value": "false"
      }

ebaauw, 2018-04-10:
Does deCONZ issue a web socket notification when the light is switched off?

Krocko, 2018-04-10:
How to check this?

ebaauw, 2018-04-10:
Through `dc_eventlog`, which should be installed with homebridge-hue.

Krocko, 2018-04-10:
Yes, i think so. 

`pi@raspberrypi:~ $ dc_eventlog
[Tue Apr 10 2018 20:00:54] listening on ws://localhost:443
[Tue Apr 10 2018 20:00:54] /sensors/8/state: {
  "lastupdated": "2018-04-10T18:00:54",
  "status": 3
}
[Tue Apr 10 2018 20:01:04] /sensors/2/state: {
  "lastupdated": "2018-04-10T18:01:04",
  "presence": true
}
[Tue Apr 10 2018 20:01:04] /sensors/7/state: {
  "lastupdated": "2018-04-10T18:01:04",
  "status": 1
}
[Tue Apr 10 2018 20:01:04] /sensors/7/state: {
  "lastupdated": "2018-04-10T18:01:04",
  "status": 2
}
[Tue Apr 10 2018 20:01:04] /lights/5/state: {
  "on": true
}
[Tue Apr 10 2018 20:01:04] /groups/5/state: {
  "any_on": true
}
[Tue Apr 10 2018 20:01:04] /groups/8/state: {
  "any_on": true
}
[Tue Apr 10 2018 20:01:09] /sensors/1/state: {
  "dark": false,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:09",
  "lightlevel": 5986,
  "lux": 4
}
[Tue Apr 10 2018 20:01:14] /sensors/1/state: {
  "dark": false,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:14",
  "lightlevel": 5864,
  "lux": 4
}
[Tue Apr 10 2018 20:01:15] /lights/5/state: {
  "on": false
}
[Tue Apr 10 2018 20:01:15] /groups/5/state: {
  "any_on": false
}
[Tue Apr 10 2018 20:01:15] /groups/8/state: {
  "any_on": false
}
[Tue Apr 10 2018 20:01:19] /sensors/1/state: {
  "dark": true,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:19",
  "lightlevel": 0,
  "lux": 0
}
[Tue Apr 10 2018 20:01:21] /sensors/6/state: {
  "lastupdated": "2018-04-10T18:01:21",
  "temperature": 2164
}
[Tue Apr 10 2018 20:01:23] /sensors/4/state: {
  "dark": true,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:23",
  "lightlevel": 423,
  "lux": 1
}
[Tue Apr 10 2018 20:01:24] /sensors/2/state: {
  "lastupdated": "2018-04-10T18:01:24",
  "presence": false
}
[Tue Apr 10 2018 20:01:24] /sensors/7/state: {
  "lastupdated": "2018-04-10T18:01:24",
  "status": 0
}
[Tue Apr 10 2018 20:01:24] /sensors/8/state: {
  "lastupdated": "2018-04-10T18:01:24",
  "status": 0
}
[Tue Apr 10 2018 20:01:42] /sensors/11/config: {
  "battery": 100,
  "on": true,
  "reachable": true,
  "temperature": 2000
}
[Tue Apr 10 2018 20:01:42] /sensors/10/config: {
  "battery": 100,
  "on": true,
  "reachable": true,
  "temperature": 2000
}
[Tue Apr 10 2018 20:01:48] /sensors/3/state: {
  "lastupdated": "2018-04-10T18:01:48",
  "temperature": 2114
}
[Tue Apr 10 2018 20:01:49] /sensors/1/state: {
  "dark": true,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:49",
  "lightlevel": 0,
  "lux": 0
}
[Tue Apr 10 2018 20:01:53] /sensors/4/state: {
  "dark": true,
  "daylight": false,
  "lastupdated": "2018-04-10T18:01:53",
  "lightlevel": 423,
  "lux": 1
}

`

manup, 2018-04-23:
Can you post the full rule and a short note what should happen when?

Krocko, 2018-04-23:
Here are  a set of rules:

```
"1": {
    "actions": [
      {
        "address": "/sensors/7/state",
        "body": {
          "status": 1
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "dx"
      },
      {
        "address": "/sensors/7/state/status",
        "operator": "eq",
        "value": "0"
      },
      {
        "address": "/sensors/1/state/dark",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/lights/5/state/on",
        "operator": "eq",
        "value": "false"
      }
    ],
    "name": "Flur Bewegung",
    "periodic": 0,
    "status": "enabled"
  },
  "2": {
    "actions": [
      {
        "address": "/sensors/7/state",
        "body": {
          "status": 1
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/sensors/1/state/dark",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/sensors/1/state/dark",
        "operator": "dx"
      },
      {
        "address": "/sensors/7/state/status",
[... 229 lines cut ...]
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/lights/5/state/on",
        "operator": "eq",
        "value": "false"
      },
      {
        "address": "/sensors/2/state/presence",
        "operator": "eq",
        "value": "false"
      }
    ],
    "name": "Bewegung Flur zurücksetzen",
    "periodic": 0,
    "status": "enabled"
  }
```

If i manually turn off the light, rule number 9 should fire, but the countdown from rules 5 and 7 goes on.

I have more of such set of rules and all stop working from time to time.

Krocko, 2018-04-23:
The following rule also don't work anymore:

```
{
  "actions": [
    {
      "address": "/sensors/36/state",
      "body": {
        "status": 0
      },
      "method": "PUT"
    },
    {
      "address": "/sensors/37/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    },
    {
      "address": "/sensors/38/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/lights/7/state/on",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2018-01-31T15:36:47",
  "etag": "dc4e5ae20ba946b2b36749137641adae",
  "lasttriggered": "none",
  "name": "Schranklampe wiederherstellen",
  "owner": "90EB5834A7",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 206
}
```

If i switch off the light, sensors don't reset their states.

manup, 2018-04-23:
Interesting I'll try to recreate the conditions.

One question, how do you turn off the light &mdash; REST-API (which request) or Phoscon App (group, single light control)?

Krocko, 2018-04-23:
I turn off the light with HomeKit (homebridge-hue plugin).

Krocko, 2018-04-23:
I added now 
 ```
{
        "address": "/lights/7/state/on",
        "operator": "dx"
      }
```

to the condition and the rule is working again. But for how long...

Krocko, 2018-04-24:
Same problem again. This rules stop working. If i modify this rules (change a condition), the rule is working. After this i can change back the condition and the rule is still working for some hours.
@manup, @ebaauw have you any idea?

manup, 2018-04-24:
With the `dx` condition the rule should trigger whenever the lights turns from on to off, looks ok.
I'm not sure if the related event is emitted in any case the lights state.on changes (there are multiple ways to do this).

However can you confirm that the rules `timestriggered` counter doesn't increase then the rule seemingly stops working? You need to do a GET request to the rule to see that.

Krocko, 2018-04-25:
If the rules stop working, `"lasttriggered"` is `"none"` and `timestriggered`doesn‘t increase.

ebaauw, 2018-04-25:
I don’t have any rules with a condition on a light, so just a wild guess: when are the trigger conditions “indexed”?  Could it be the light resource is re-created too late after restarting deCONZ?

Krocko, 2018-04-25:
This is a good point. Because `"lasttriggered"` becomes `"none"` after a restart of deCONZ.

manup, 2018-04-25:
>I don’t have any rules with a condition on a light, so just a wild guess: when are the trigger conditions “indexed”? Could it be the light resource is re-created too late after restarting deCONZ?

Might be I need to check indexing. Also the delayed light resource creation is on my screen so that they will be created like sensors right after startup.

I don't have any light rules either, I'm solving the effect of — reset state when lights go off — which via group.any_on.

Krocko, 2018-04-25:
In this groups are smart plugs also for radio, microwave, fan. So a condition group.any_on makes no sense. Or i must create groups with only one light.

ebaauw, 2018-04-25:
And cross your fingers that your plug does support groups...

Krocko, 2018-04-25:
It is definitely after a restarted from deCONZ. Then the rules stop working. If i change only one of this rules, the other rules are working again.

ebaauw, 2018-04-25:
> If i change only one of this rules, the other rules are working again.

That probably triggers the re-indexing of the trigger conditions.

Krocko, 2018-04-26:
Ok. Is there any way to fix this? Or is there any workaround?

manup, 2018-04-26:
Should be fixed in the next version 2.05.25, re-indexing must be called from the light loading function.

Krocko, 2018-04-26:
Thank you very much @manup

Krocko, 2018-04-27:
Works now. Good work.

closed 2018-04-27 by Krocko