dresden-elektronik/deconz-rest-plugin #531: ZHAPower & ZHAConsumption - documentation available?

jo-me, 2018-04-12:
Hi,

adding a smart plug (e.g. Heiman SmartPlug) results in two new sensors in deconz, one of type _ZHAConsumption_ and one of type _ZHAPower_.

Is there documentation available somewhere that indicates how the values can be interpreted?

ZHAConsumption looks like this:
```
"state": {
      "consumption": 10,
      "lastupdated": "2018-04-12T18:14:05",
      "power": 23
  },
```

ZHAPower state looks like this:
```
"state": {
      "current": 14,
      "lastupdated": "2018-04-12T18:14:04",
      "power": 23,
      "voltage": 233
 }
```

Could you please correct and complete the description if it is not available elsewhere?

- ZHAConsumption.consumption = energy consumption in kWh, summed up in the plug itself not in deconz
- ZHAConsumption.power = ?
- ZHAPower.current = ?  - i guess the current current in A or mA, but the value does not make sense to me
- ZHAPower.power = current power usage in Watts
- ZHAPower.voltage = current voltage in Volts

Thanks
Jochen

ebaauw, 2018-04-12:
All attributes are as reported by the plug, no calculations by deCONZ, except scaling to tye right units:
- Consumption in kWh;
- Power in W;
- Voltage in V;
- Current in 0.1A.

Consumption is indeed summed by the plug itself.  It’s reset to 0 when you reset the plug.  Power is actually reported in both clusters, hence it appears in both resources.

jo-me, 2018-04-14:
Thank you for clarifying this!

ebaauw, 2018-04-14:
You’re welcome, but I erred: consumption is in 0.001 kWh (or if you prefer in Wh).

closed 2019-03-30 by stale[bot]