dresden-elektronik/deconz-rest-plugin #534: Support for ubisys Shutter Control J1?

felixstorm, 2018-04-13:
Hi @manup, hi @ebaauw,

I saw from #404, #269 and #209 that you're currently thinking about opening up deCONZ for other actuators besides lights. I have quite a few ubisys J1 that I have paired with deCONZ but currently I can't really control them in any way. It would be great if general support for them could be included as well - so if I can be of any help with this please get in contact with me.

And many thanks for the great work on deCONZ!

Felix

felixstorm, 2018-04-13:
They have pretty detailed ZigBee information available here:  
http://ubisys.de/downloads/ubisys-j1-technical-reference.pdf

A few months ago I had tried to add their configuration attributes to general.xml, but as they use pretty tricky ZigBee attribute type configurations especially to configure the inputs on their devices, this did not really work out.

felixstorm, 2018-04-13:
Sorry to spam you with multiple comments, but another thing just came to my mind: A functionality to directly bind ZigBee switch inputs (like the C4s) to the J1s on the ZigBee level would also be helpful for me, but I am not sure if this would also be within the scope of #404, #269 etc.

schneidermarc, 2018-04-16:
Hi Felix,

take a look at my zcl file:
https://gist.github.com/schneidermarc/0a5362d4bd60d2301c536e37a9497e1c

My J1 works petty well on the DeCONZ GUI.  I am currently in contact with the support to clarify the possibilities to link the zcl profiles with theREST API.

Marc

felixstorm, 2018-04-16:
Great, thanks @schneidermarc - I will definitely try it out for control as soon as I have a GUI again. Currently, my deCONZ is running headlessly in a docker container, but should probably soon get an X11 UI again (see https://github.com/marthoc/docker-deconz/issues/15).

When I tried customizing the zcl file myself a few months ago, I wanted to configure buttons and calibration first and this has been implemented by ubisys with rather complicated zcl types (InputActions is an array of attributes with subtype raw binary data) and also requires writing to attributes with the "manufacturer specific" bit set - both of which deCONZ can't or at least could not do at that time, so I stopped looking in this direction. Back then I managed to do these tasks with the help of an XBee module, an abandoned ZCL library for C# and a few lines of code, but it was far from being pretty.

As I use Home Assistant now and temperatures get warmer in the norther hemisphere, automatic shutter control for sun protection has now moved up quite high on my priorities list and seeing #404 etc., I drew new hope ;-)  
I also thought about going away from deCONZ and only using [ZiGate](https://github.com/elric91/homeassistant_zigate) or my existing XBee module with Home Assistant instead, but as deCONZ is being maintained so well, I would really like to stay with it. But for that I would at least at some point in time need to also be able to control and manage (group/scene assignments) the J1s through the REST API (even if it was with raw ZCL frames).

Felix

manup, 2018-04-16:
With the current work on various ubisys devices (D1, S1, S2 and C4) and the manufacturer specific *Device Setup Cluster (0xFC00)* the J1 should follow as logical step. It's one of the devices which should be placed in the new */devices* (?) endpoint.

felixstorm, 2018-04-16:
Thank you for the info @manup! I definitely don't want to push here - knowing that there's a realistic chance for support at all I'm happy to wait.

And if I can be of any help just contact me - I have about 12-14 of these in use.

Felix

ma-ca, 2018-08-04:
From the deCONZ UI in the latest version 2.05.32 it is already possible to send the commands move up/down. 

I have created a PR #709 to support ubisys J1 in the rest plugin. For now it just displays the J1 in the /lights without doing anything else. The next step is to implement commands. Just let me know if I am on the right track and I will continue working on it. My goal is to use HomeKit to control the J1.

The ubisys J1 shows up in the /lights:

```
  "lights": {
...
    "7": {
      "etag": "c021a6dbdb3de3ac17a7dd508887675b",
      "hascolor": false,
      "manufacturername": "ubisys",
      "modelid": "J1 (5502)",
      "name": "Light 7",
      "state": {
        "alert": "none",
        "on": false,
        "reachable": true
      },
      "swversion": null,
      "type": "Window covering device",
      "uniqueid": "00:1f:ee:00:00:00:25:3b-01"
    },
...
```

felixstorm, 2018-08-10:
Thanks for your PR, @ma-ca - I just installed the latest version and all J1s do appear as lights!  
To me this sounds like a good first step. If On/Off could be translated to Move Up/Down, Toggle to Stop and Brightness Level and Hue to Lift and Tilt positions, then it would almost be possible to fully control the J1s ;-)

For my personal use case (controlling ubisys J1 trough Home Assistant) I found a temporary solution using modified versions of zigpy and zigpy-xbee and a custom ZHA cover component (https://github.com/felixstorm/homeassistant-config/blob/master/custom_components/cover/zha.py), but this is all far from being pretty or stable, so support within deCONZ would still be highly appreciated.

ma-ca, 2018-08-12:
@felixstorm in PR #727 I have implemented your suggestion to have on/off translate to close/open and bri to move to tilt percentage.

ebaauw, 2018-08-19:
How do you control tilt vs up/down, where the J1 only has two outputs (up and down)?

ma-ca, 2018-08-19:
The J1 can only move up/down or stop through the inputs from the buttons. **After calibrating** the J1 can also move to tilt percentage but that command is only available though ZigBee commands. During the calibration the J1 counts the steps it takes from fully closed to fully open. The window covering cluster then is able to move to tilt in percentage.

ma-ca, 2018-08-19:
> How does the J1 control tilt vs up/down, where it only has two outputs (up and down)?

I was mixing tilt and lift, so I don't know the answer to that question. I am only using lift.

ebaauw, 2018-08-19:
Yeah, from the ZigBee description, you can set what type of shutter is connected to the J1, and consequently whether it supports lift, tilt, or both.  I'm still looking to smartify my venetian blinds and was wondering what the connection from the J1 to the lift/tilt shutter would look like.

felixstorm, 2018-08-19:
I have external venetian blinds that have only one motor and therefore also just need two power lines for up and down just like a roller shutter.  

The way tilt control works mechanically is that when the motor starts turning in closing direction, at first only the blinds tilt but do not yet move down. After about 1.5 seconds, when the blinds have tilted completely, they start moving down. Same goes for the opposite direction: When the motor starts turning in the upwards/opening direction, at first only the blinds tilt and then start moving upwards thereafter. I‘d say that this is pretty standard here in Germany for most motorized venetion blinds (we also have internal ones at the office that work the same way).  
When the blinds are fully open and the J1 is told to move to e.g. 30% lift and 50% tilt (perfect for sun protection but you still get enough light and can easily see through to the outside), the motor will turn in closing direction until 30% lift is reached and then turn backwards again for about 750ms to open the blinds as requested without moving them upwards again.

The time it takes to fully tilt the blinds from 0 to 100% cannot be detected automatically by the J1 (as for lift), but must be manually configured through the attributes LiftToTiltTransitionSteps and LiftToTiltTransitionSteps2 in the window covering cluster.

Also, AFAIK there is no single ZigBee command to control both values at the same time, so I move to the desired tilt position first (as this will always be completed within 2-3 seconds), wait 5 seconds and then send the command to move to the desired lift position, and the J1 will automatically turn the blinds to the tilt position requested before after having reached the correct lift position (currently using a custom cover component based on zha/zigpy controlling an additional XBee S2C module in Home Assistant).

felixstorm, 2018-11-04:
Thanks @ma-ca for implementing this!

closed 2018-11-04 by felixstorm

overdrive69, 2019-12-22:
Hi, how do you control the j1?

martingruening, 2020-04-17:
This sounds like the Ubisys J1 works with deconz, but it is not listed in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices - are there still issues that prevent it to be marked as 'supported'?

cosmo84, 2020-12-06:
I follow Martin's question. Any reasons why the J1 is not marked as supported? Any feedback is appreciated, since I need to decide to buy the J1 or the shutter switch from Legrand. 

Thanks a lot.

SwoopX, 2020-12-06:
Hm, not mentioned? Than that is just accidentially.

No worries, got one and it's running just fine

cosmo84, 2020-12-06:
Thanks for the feedback!