dresden-elektronik/deconz-rest-plugin #536: Is "Wireless Light Control" deprecated?

mysteryan, 2018-04-13:
I noticed my lights stopped showing up in Wireless Light Control since 2.05.18 - 2.05.21, but they show up in PWA and third party apps. Is the webapp being phased out?

manup, 2018-04-13:
Other than small fixes it is not actively maintained anymore but lights should work fine.

Since 2.05.20 it can be opened from the Phoscon App `Menu > Help > Old WebApp Version` as well as from the deCONZ GUI via the WebApp button.

manup, 2018-04-15:
Interesting I'll check this out, have an "old"  iOS 10 iPhone 5 laying around here.

manup, 2018-04-15:
Inbetween it should be possible to open the old WebApp on these devices by directly opening http://<ipaddress>/login.html

mysteryan, 2018-04-19:
Manually navigating to login.html was one of the first things I tried thinking that it was a mismatch of credentials.  Upon logging in, it takes me to the same Wireless Light Control webapp that doesn't have any data populated in it.

Specifically, I get this issue on an iPhone 7 on iOS 10.1.1, iPad Pro, iPad Air, iPhone 6+ on iOS 10.2.1, but the webapp displays properly on an iPad 4 on iOS 10.3.3.  Were you able to replicate this issue on your iPhone 5?

This isn't that big of issue since I didn't see anyone else encountering the same issue and PWA or alternative apps can always be used on the affected devices.  I just preferred the old webapp cause it provides quicker access to all lights, their states and their scenes with minimal navigation through submenus and tabs.

manup, 2018-11-23:
This should work with recent versions.

closed 2018-11-23 by manup