dresden-elektronik/deconz-rest-plugin #541: Firmware Raspbee does not work with PI 3b+

QHose, 2018-04-15:
hi,

I just tested and both the beta as well as the stable [firmware ](https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/raspbee-gw-sd-card-image/) don't work. When I download the image and write it to SD card. the PI flashes 4 times slow 4 times fast.

This raspbian does boot the machine https://www.raspberrypi.org/downloads/raspbian/
But now I can't make use of all the preinstalled software that the dresden image provides, any ideas?

QempZor, 2018-05-30:
Im also trying to install DeCONZ on my new RPi3b+
However, im also running in some problems.
Able to boot and start DeCONZ, but only showing a Daylight node..

willsy555, 2018-10-07:
I had the same issue. Only get the daylight node on a rpi3b+ and works on a rpi2. Is 3b+ support expected?

jjlawren, 2019-02-03:
Is this still the case? I was about to deploy a new RaspBee on a 3B+ but now I’m worried it won’t work.

jjlawren, 2019-02-09:
Works fine, just need these entries in the `config.txt`:
```
enable_uart=1
dtoverlay=pi3-miniuart-bt
```
I didn't see this in the documentation. Worth adding? And maybe update supported models here: https://www.dresden-elektronik.de/funktechnik/service/support/faq/wireless-light-control/?L=1

kallipso, 2019-04-20:
Solved  rpi3+
i have the same problem i change  "/dev/ttyAMA0" to "/dev/ttyS0"

manup, 2019-04-20:
>Solved rpi3+
>i have the same problem i change "/dev/ttyAMA0" to "/dev/ttyS0"

When the serial port is configured as in step 1) it right device will be /dev/ttyS0

https://phoscon.de/raspbee/install#raspbian

Can also be verified with:

    ls -l /dev/serial0

Which should be a symlink poining to `/dev/ttyS0`, before configuration it points to `/dev/ttyAMA0`.

closed 2019-08-25 by stale[bot]