dresden-elektronik/deconz-rest-plugin #542: deCONZ REST-API Development on Windows?

felixstorm, 2018-04-15:
I would be interested in trying to extend the REST-API plugin, but due to several reasons my main working machine is a Windows 10 machine and deCONZ runs on a Raspberry Pi.

As the Pi hardly seems feasible as a development machine, is there an easy way to compile the plugin on Windows? In the readme compiling it on Linux is documented, but I could not find anything about doing it on Windows.

Or has anybody yet tried to work with deCONZ in the [Windows Subsystem for Linux](https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux)?  
It's included with Windows 10, comes with a full Ubuntu userland and runs native Linux ELF binaries, but only 64 bit.

Thanks,  
Felix

felixstorm, 2018-04-16:
A short update: I was actually able to start deCONZ within WSL (even with the X11 GUI), but unfortunately deCONZ does not detect the ConBee even though it's available within WSL as /dev/ttyS6.

@manup - Is there a way to force deCONZ to a specific serial port? My guess would be that deCONZ does not detect the ConBee because it is only available as a plain serial port, but not as an USB device.

Thanks,  
Felix

manup, 2018-04-16:
>@manup - Is there a way to force deCONZ to a specific serial port? My guess would be that deCONZ does not detect the ConBee because it is only available as a plain serial port, but not as an USB device.

Not yet but it is planned for the next days, since it is also required for easier docker image integration.

felixstorm, 2018-04-16:
Great, thank you!

felixstorm, 2018-11-04:
It's available for some time now by using the switch --dev when starting deCONZ - thanks @manup !

closed 2018-11-04 by felixstorm