dresden-elektronik/deconz-rest-plugin #543: Xiaomi Aqara wall switch

farbrorlarry, 2018-04-15:
I tried to get one dual wall switch to work but it seems not working. With Phoscon App I can see the switch connected as Vendor LUMI/Product Transmitter 2-gang. But when I look at Groups it doesn't show? I'm not sure what information I need to give you but I can donate a button. I think it's faster for you than try to explain all steps to get correct information.

mathos77, 2018-04-22:
@1968-Larry 

Please see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356

@manup I suggest to close this issue.

ebaauw, 2018-04-29:
This is not an in-wall switch, wired to the electricity, but a battery-powered switch that looks like a wall switch.  Please keep this issue open.

1968-Larry, 2018-04-29:
Moving my report here:
I have upgraded to latest package available, removed the switches in pwa and database and tried to connect it back to the network. It's paired and I can see it in pwa/device/switches but I can't use it in groups. I tried to read /api/apikey/switches and it's missing.
I got this in the database:
"3549"	"5"	"lumi.sensor_86sw2 5"	"ZHASwitch"	"lumi.sensor_86sw2"	"LUMI"	"00:15:8d:00:01:e0:75:db-01-0006"	""	"{""buttonevent"":null,""lastupdated"":null}"	"{""battery"":null,""on"":true,""reachable"":true,""temperature"":null}"	"{""d"":65535,""ep"":1,""in"":[0,6],""p"":260}"	"normal"	"1"

I could give you more information if you know what to look for.

ebaauw, 2018-04-29:
That looks good to me.  Does `state.buttonevent` change value when you press a button?

These switches use attribute reporting instead of commands to change associated lights or groups.  They don’t work standalone, without a gateway.  You need gateway rules to change the lights or group when `state.buttonevent` is updated.  Consequently, `config.group` is missing, just like for my Xioami switches: `lumi.sensor_switch.aq2`, `lumi.cube`, and `lumi.ctrl_ln2.aq1`.

 Don’t think PWA yet supports creating the needed rules.

1968-Larry, 2018-04-29:
I'm not sure what to look at but I think it's always:
{"buttonevent":null,"lastupdated":"none"}
Could I add the rules in database?

ebaauw, 2018-04-29:
> I'm not sure what to look

Best use the REST API, doing a GET of /sensors/5.  The database only updates periodically and when deCONZ exits.  
Using `ph` from my [homebridge-hue-utils](https://github.com/ebaauw/homebridge-hue-utils), you'd simply do:
```
$ ph get /sensors/115
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true,
    "temperature": 2300
  },
  "ep": 1,
  "etag": "ea5af57630320ec809e4f39b75b92e92",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.sensor_switch.aq2",
  "name": "Entrance Switch",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2018-04-22T05:07:30"
  },
  "swversion": "3000-0001",
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:a6:6f:3a-01-0006"
}
```

> Could I add the rules in database?

Best create rules through the REST API.  They'll be stored in the database, but the API provides some sanity checks.

1968-Larry, 2018-04-29:
Nothing happens when I press the buttons:
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": false,
    "temperature": 2600
  },
  "ep": 1,
  "etag": "996e9ed6567e3b88da0ad87ee3f4ebce",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.sensor_86sw2",
  "name": "lumi.sensor_86sw2 5",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:e0:75:db-01-0006"
}

ebaauw, 2018-05-07:
Can some-one please post the screenshots for the `lumi.sensor_86sw1` and the `lumi.sensor_86sw2` of:
- The node with endpoints and clusters;
- The _Node Info_ panel;
- The _Cluster Info_ panel for the _Basic_ cluster.

And check which endpoints/clusters/attributes change when (single, double, triple, long) pressing the buttons?

And list what `/sensors` resource(s) is/are created in the REST API?

Thanks.

alexvandervegt, 2018-05-08:
@ebaauw I believe this information is in issue:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356

ebaauw, 2018-05-08:
No, those are the _wired_ in-wall switches (`lumi.ctrl_neutral` and `lumi.ctrl_ln`), not the wireless switches that look like wall switches.

1968-Larry, 2018-05-08:
I hope this is right.

[image]
[image]
[image]

ebaauw, 2018-05-08:
Thanks, @1968-Larry 

Could you please:
- Read the attributes in the _Basic_ cluster?  You might need to press the reset button (if present) or regular button on the switch to make sure it’s awake;
- Check which of the _OnOff_ clusters _On_ attribute changes to true on pressing which button or maybe both buttons?  Looking at the code, deCONZ should handle these.  Did these switches ever work?
- Check wether _Current Value_ in the _Multistate Input_ cluster and in the _Analog Input_ cluster changes on pressing a button?

ebaauw, 2018-05-08:
Reference: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/165#issuecomment-348329845

Same (?) switch, but different clusters and device IDs?!

1968-Larry, 2018-05-08:
@ebaauw will try to night, I can't reach the button from here :) If you send me your address I could send one to you.

ebaauw, 2018-05-08:
> If you send me your address I could send one to you.

If you could loan me one, that would be brilliant.  Can I contact you by email, to send my address?

1968-Larry, 2018-05-08:
I added a public email to my github account.

ebaauw, 2018-05-11:
Hm.  Re-installed v0.05.25.  Deleted sensor resource from database, reset sensor, deleted node from deCONZ GUI and paired the sensor again.  This time, it only shows _Basic_ and _OnOff_ clusters on endpoint 01, and _OnOff_ on endpoint 03.  When pressing the left button, endpoint 02 appears, also with only _OnOff_ cluster.  Device ID is now 0xffff for all endpoints.

Again, after pressing a button (to wake the switch) and manually reading the _Basic_ cluster attributes, the ZHASwitch sensor resource is created, and, again, I'm getting buttonevents.

ebaauw, 2018-05-11:
Take 3.  This time, I'm getting some light resources (?!).  Probably because of the 0xffff device types deCONZ confuses it for a `lumi.ctrl_neutral`.
```
{
  "3": {
    "etag": "53f2002e30c1c7aac7f6d58432c6cd49",
    "hascolor": false,
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "Light 3",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "00:15:8d:00:01:e0:75:d6-03"
  },
  "4": {
    "etag": "53f2002e30c1c7aac7f6d58432c6cd49",
    "hascolor": false,
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "Light 4",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "00:15:8d:00:01:e0:75:d6-02"
  }
}
```
No matter what I try, the _Basic_ attributes won't be read.  Somehow, deCONZ got the _Model Identifier_ already, and created the sensor resource (note: no `swversion` this time):
```
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true,
    "temperature": 2900
  },
  "ep": 1,
  "etag": "f8f7e6fa2091cb647bde1845aeca3089",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.sensor_86sw2",
  "name": "lumi.sensor_86sw2 7",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2018-05-11T15:01:50"
  },
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:e0:75:d6-01-0006"
}
```
Again, I'm receiving buttonevents and Xiaomi magic reports.

ebaauw, 2018-05-11:
Take 4.  Reset the switch, removed the node, shutdown deCONZ, reset the RaspBee, power-cycled the only other device (on my test network), removed the light and sensor resources from the database, and once again paired the switch.  This time is comes back with the full endpoints and clusters, but the _OnOff_ clusters only appear on the corresponding button action.  Device IDs on endpoints 02 and 03 are 0x5f02 and 0x5f03, so no lights this time.  Again the sensor resource is created and I do get buttonevents.

@manup, I don't know how the node's endpoints and clusters can differ, but it would seem that the light discovery and the sensor discovery interfere with each other.  I think the current implementation of light discovery for battery-powered end devices is cursed (mea culpa) and we better create the light resources from addSensorNode(), when we know the _Model Id_.  We could simply whitelist the endpoints and as added benefit, update the manufacturername and modelid the same way we do for sensor resources.

saurabh984, 2018-05-22:
Hi Erik,

Let me know if you need any help debugging this. WXKG02LM & WXKG03LM both work for me using deCONZ v2.05.27

It took several attempts to get them to join, report correct attributes, device ids and be reported by the REST API. I found that resetting them first by pressing the button for >5 seconds after removing and reinserting the battery had better results.

After doing the reset cycle I added them again by opening the network and pressing the button for 5 seconds.

WXKG02LM hasn't had any config reports yet so getting battery info etc as null. It was reporting when I paired it the first time.

Another strange observation - A brand new unpaired switch seems to work the very first time (after pressing the buttons to get the ZHASwitch Resource created). If any attempt to repair or reset then these are an absolute nightmare!

Hope it helps!

fillwe, 2018-09-25:
Hey guys i just bought both a WXKG02LM & WXKG03LM. After 4 hours i still havent been able to add any of them to Deconz. I have tried your method with restoring saurabh984, replacing the battery nothing :( Any time i hold in the button, both leds just starts flashing 6 times, then nothing, have waited for it to pair but with no result. Any ideas?:(

zilluss, 2018-10-22:
@fillwe Have you been able to connect the switch? I'm on the latest version and can't connect it either. I'm currently considering switching to a CC2531/Zigbee2MQTT based system but I'm not sure if I'm simply doing something wrong while trying to connect the switch.

closed 2019-02-26 by stale[bot]