dresden-elektronik/deconz-rest-plugin #544: How to add xiaomi devices to phoscon app?

QHose, 2018-04-16:
hi all,

I see I can add multiple brands, but xiaomi is not listed in the phoscon app
* search lights is generic and will find all lights of all brands I guess
* buttons: Xiaomi or "other" is not available, so how to add?
* sensors: other is availalbe, I did a scan, but nothing appeared. 

Also I wondered, sometimes I see people also use the deconz app directly, when to use that versus phoscon?

Setup: new raspberry pi 3b+ with latest deconz

thanks!

gjong, 2018-04-16:
For the sensors it is easy. Select 'Other' and start the pairing process by pressing the button on the Xiaomi sensor for 4 seconds. It will pair some seconds after that. Do note that in my experience it's a bit hit and miss and you might have to try several times. I've successfully done this with door sensors and temperature sensors.

The Xiaomi lights are WIFI enabled lights, so I don't see how you would connect them to deCONZ (perhaps you found some Zigbee enabled once, but I haven't so far).

nerdosity, 2018-04-16:
I added a couple of Xiaomi door opening sensor.
One Aqara, one Mijia version.
It worked, but:

- no battery status, that's REALLY bad.
- if I try to add my switch wireless button, it let me add it, but it doesn't appear on the gui.

[image]
[image]

manup, 2018-04-16:
Battery status needs up to 1 hour until reported (Xiaomi devices send these reports hourly at least Aqara version).

Whats the modelid of your switch? Not all switches are displayed in the Phoscon App yet.

nerdosity, 2018-04-16:
Added in 16/4/2018, 16:42:52.
Still no battery status.
How battery will be reported? I have a plugin that "automagically" create battery states, and it worked for Xiaomi battery status.

About the switch, it's the WXKG01LM. With Xiaomi Gateway, I was able to use as a press, long press, double press.
How can I use it now?

manup, 2018-04-16:
Should look like this:

[image]

>About the switch, it's the WXKG01LM. With Xiaomi Gateway, I was able to use as a press, long press, double press.
How can I use it now?

Currently only by using the REST-API, later on there will be a switch editor in the Phoscon App where the buttons can be assigned to various actions.

nerdosity, 2018-04-21:
There is some big issue with battery status reporting.
I added two [temperature sensors from Xiaomi](https://www.banggood.com/New-Arrival-Original-Xiaomi-Mini-Smart-Home-Temperature-and-Humidity-Sensor-White-p-1046061.html?cur_warehouse=CN) (the Mijia version, not the Aqara one with pressure)

[image]

[image]

I added them yesterday around 7PM. Now after more than 12 hours, still no battery status.
How can we fix this issue? I can't drop and readd my sensors when I want a battery report just like that, it can't be trusted!
With the official gateway + home assistant, the battery status appears ALWAYS the first time the door sensor change status. After a reboot, a battery or two could be missing... but opening a door, or using a switches, ALWAYS gave me a battery refresh.

But this way, I can't trust at all the battery percentage... It looks like it's refreshing randomly, WHEN it refreshes.
The temp sensor doesn't even get a battery status after more than 12 hours.

I think that I am going to return the Raspbee, price is good (comparable to just any gateway) but it doesn't look like it can be trusted for battery status. And for wireless sensors, that's too important to be "random".

QHose, 2018-04-21:
For others: You have to add one sensor add at time, press search, reset your sensor with a paperclip (you can turn it, to remove it from the sticker (the plastic with glue), bring it close to the conbee, and wait a couple of secs, then redo for other sensors.

Paalap, 2018-08-13:
Hi I have also the issue with adding WXKG01LM (mi smart switch or button) to Phoscon app. It is actually connected (I can access it via iobroker) but not visible in the Phoscon GUI. Why or how can I fix it?

Thanks

Paalap, 2018-08-13:
Cool, thanks!

SkinnyNorris123, 2018-08-14:
Hi, I`m trying to add the Aqara Water Sensor to the Phoscon App without success. Is says "added", but I can not see it.
Can you pls comment if this Sensor is supported or will be in the future ?

manup, 2018-08-14:
The Water sensor is currently not visible in the Phoscon App, only in the via REST-API. With help of further systems like Home Assistant, Homebridge-hue, etc the sensor can be visualized and automated.

SkinnyNorris123, 2018-08-14:
I`m running FHEM, can you maybe point me to some more details ? I can see the Sensor there, but only the battery not the "Water value"

manup, 2018-08-14:
Can you provide the output delivered by the REST-API?

ebaauw, 2018-08-14:
It should look like:
```json
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true,
    "temperature": 2700
  },
  "ep": 1,
  "etag": "d777996c63684ab53672acc2108c02ce",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_wleak.aq1",
  "name": "Boiler Leak",
  "state": {
    "lastupdated": "2018-08-14T14:33:55",
    "lowbattery": false,
    "tampered": false,
    "water": false
  },
  "swversion": "3000-0001",
  "type": "ZHAWater",
  "uniqueid": "00:15:8d:00:02:33:f2:90-01-0500"
}
```

SkinnyNorris123, 2018-08-15:
Thanks for the example..... The API shows the right value:
`{
"config":{
"battery":98,
"on":true,
"reachable":true,
"temperature":2500},
"ep":1,
"etag":"8a56513f8b422a256dcdb04e240c9dda",
"manufacturername":"LUMI",
"modelid":"lumi.sensor_wleak.aq1",
"name":"lumi.sensor_wleak.aq1",
"state":{"lastupdated":"2018-08-15T09:51:53",
"lowbattery":false,
"tampered":false,
"water":false
},
"type":"ZHAWater",
"uniqueid":"00:15:8d:00:02:12:f2:9f-01-0500"
}
	
`
So looks like I need to work on fhem to get this working.
Thank you very much

Uakkauakkafo, 2018-10-10:
hello,
I hope this is the rigth place:

I try to link the  Water sensor from Xiaomi (sensor_wleak.aq1) to zigbee net via deconz Phoscon App or WebApp but nothing, in the gui I see this:

[image]

I also used Pstman to search the sensor but I see only the other.

can someone help me?

thanks

manup, 2018-10-10:
Water leak sensors aren't visible in the Phoscon App yet, the issue is on the todo list but will take a few releases.

Uakkauakkafo, 2018-10-10:
But in July I had a different version od deCONZ and I was able to saw it and link in Home Assistant.

Now I don't remember what version I had, the actual one is the 2.05.39, so is a problem of actual version?

Uakkauakkafo, 2018-10-10:
I also try to see it via Postman (Rest-API) with the following command: 

IPconbee:8080/api/TOKEN/sensors

but nothing, I see for example the Xiaomi Temp sensor and the Xiaomi Cube:

    },
    "5": {
        "config": {
            "battery": 95,
            "on": true,
            "reachable": false,
            "temperature": 2400
        },
        "ep": 2,
        "etag": "4bad52559b34c1d9a3aeeed36afba702",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube",
        "name": "lumi.sensor_cube",
        "state": {
            "buttonevent": 7007,
            "lastupdated": "2018-10-10T14:39:23"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:03:8d:a4-02-0012"
    },
    "6": {
        "config": {
            "battery": 91,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "3806fd3f4f9b213cb9356a1c1da99bfc",
        "manufacturername": "LUMI",
        "modelid": "lumi.sensor_ht",
        "name": "Sensore Cameretta",
        "state": {
            "lastupdated": "2018-10-10T14:19:21",
            "temperature": 2294
        },
        "type": "ZHATemperature",
        "uniqueid": "00:15:8d:00:02:0f:f5:fa-01-0402"

but nothing related to the Water sensor.

Any help or suggestion to see it via REST-API?

I'm new of that argument, not expert, is the first time.

thanks.

manup, 2018-10-21:
I guess we need to collect more model specific data for the various Xiaomi devices and versions. Alone the one button switches are available in three (?) different versions with unique model identifiers and capabilities albeit the switches look identical.

Good place to collect device specific details should be our Wiki, I'll see to create some sub pages for the devices.

Can you please provide the details for WXKG02LM as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Uakkauakkafo, 2018-10-24:
I received a new sensor (leak sensor), I'm able to see it via Postman:

    "12": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 3000
        },
        "ep": 1,
        "etag": "55affd1df74ffb63edaae0a2d983cb01",
        "manufacturername": "LUMI",
        "modelid": "lumi.sensor_wleak.aq1",
        "name": "lumi.sensor_wleak.aq1",
        "state": {
            "lastupdated": "2018-10-24T14:09:06",
            "lowbattery": false,
            "tampered": false,
            "water": false
        },
        "type": "ZHAWater",
        "uniqueid": "00:15:8d:00:02:6e:26:33-01-0500"

and also via deCONZ:

[image]

Any suggestion?

thanks a lot

QHose, 2018-10-25:
hi,

what I found is the most easy solution:

- just buy a xiaomi gateway (now also with homekit support for 34 euro on ali express

Just use the xiaomi plugin in homebridge, and everything works brilliantly, and less depedencies, and a simplier more robust environment.

wuast94, 2018-11-02:
i run phoscon/deconz in a docker container and can´t add the aqara water leaking sensor. it is supportet in wiki but phoscon can´t find them.

closed 2019-03-10 by stale[bot]