dresden-elektronik/deconz-rest-plugin #545: Xiaomi Aqara/Mijia Zigbee Wireless Smart Switch

nerdosity, 2018-04-16:
Hi,

I have one of these buttons, Mijia version. 
I can add it as a sensor, but after that it disappears. 
I can't add it as a switch, since it miss "Other" selection, like on Sensors.

How can I add it? I use my Raspbee headless, so I just use Phoscon app to manage it.

phithor, 2018-04-18:
It is most likely added if you got the confirmation. I can currently only be used with the REST-API

manup, 2018-04-20:
Agree, we are aware of the issue and it is already on the todo list.

nerdosity, 2019-03-25:
And that's why my Raspbee went back to the seller.

manup, 2019-03-25:
The round Xiaomi button was added in 2.05.57 and is supported in the switch editor.
Meanwhile the switches page also has an add "Other" option in the wizard.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_57

deven2020, 2019-07-23:
Hi I just purchased Xiaomi WXKG12LM and try to integrate using conbee 2. Phoscon unable to discover the device. i used all 3 - Light, Sensor and Switch. Any idea/Help please.

closed 2019-11-27 by stale[bot]