dresden-elektronik/deconz-rest-plugin #547: Segmentation Fault after Power Down

kropveld, 2018-04-17:
deCONZ was running without any issues on my Raspberry Pi 3 B for many months, until I recently had a sudden power off. When I try to start deconz I get a 'segmentation fault'.
Since then I did a complete raspbian upgrade, and upgraded to the latest deb-file (2.05.20)

```
pi@stx:~ $ sudo systemctl start deconz
pi@stx:~ $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/etc/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2018-04-17 12:10:12 CEST; 7s ago
  Process: 15161 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=killed, signal=SEGV)
 Main PID: 15161 (code=killed, signal=SEGV)

apr 17 12:10:12 stx systemd[1]: deconz.service: Unit entered failed state.
apr 17 12:10:12 stx systemd[1]: deconz.service: Failed with result 'signal'.
apr 17 12:10:12 stx systemd[1]: deconz.service: Service hold-off time over, scheduling restart.
apr 17 12:10:12 stx systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
apr 17 12:10:12 stx systemd[1]: deconz.service: Start request repeated too quickly.
apr 17 12:10:12 stx systemd[1]: Failed to start deCONZ: ZigBee gateway -- REST API.
apr 17 12:10:12 stx systemd[1]: deconz.service: Unit entered failed state.
apr 17 12:10:12 stx systemd[1]: deconz.service: Failed with result 'signal'.
pi@stx:~ $ ps -ef | grep -i conz
root     15002     1  0 12:10 ?        00:00:00 /bin/bash /usr/bin/deCONZ-update2.sh
root     15038     1  0 12:10 ?        00:00:00 /bin/bash /usr/bin/deCONZ-WIFI.sh
pi       15263 12754  0 12:10 pts/1    00:00:00 grep --color=auto -i conz
pi@stx:~ $ /usr/bin/deCONZ
Segmentatiefout
pi@stx:~ $ 

```
I also checked the SD-card for any errors with fsck
I also reinserted the conBee USB-stick, and checked with dmesg that this was recognized.

Any hints?

manup, 2018-04-17:
Can you please start deCONZ with:

`deCONZ -platform minimal --http-port=80 --dbg-error=1 --dbg-info=2`

It might provide more details where the crash happens.

kropveld, 2018-04-17:
```
pi@stx:~ $ deCONZ -platform minimal --http-port=80 --dbg-error=1 --dbg-info=2
Segmentatiefout
pi@stx:~ $ deCONZ --help
Segmentatiefout
pi@stx:~ $ deCONZ -h
Segmentatiefout
```
This is what I read in the syslog

 `deCONZ-WIFI.sh[348]: command failed: Device or resource busy (-16)`

manup, 2018-04-23:
Sorry for the late response, did you also try deCONZ version 2.05.21?

If it still happens we need to make a coredump to figure out what happens.

kropveld, 2018-04-24:
While installing 2.05.21 I noticed that some other tools generate Segment Faults. Even some commands. That makes me conclude something else in the system is corrupted. I am not wasting more time to find out what, I have decided to do a completely new install. Issue can be closed :(

closed 2018-04-24 by kropveld