dresden-elektronik/deconz-rest-plugin #548: NYCE Ceiling Motion Sensor

jeubanks, 2018-04-17:
Any chance the NYCE Ceiling motion sensor will be supported soon?

On Windows with deconz loaded in the screen sometimes the sensor will show up as a node and sometimes it won't.  Within the REST api nothing is every seen/presented.

On Linux with the REST api nothing is detected.

simplextech, 2018-09-21:
Is there documentation or a process for getting devices supported?

wvuyk, 2018-09-21:
@manup, it could be very helpful I think to have a page or documentation how to request support for new devices. I am getting regular requests on how to get a sensor supported. this NYCE motion sensor is one of them and also the request for #792, the smarteinit plug.

When there is a guide, I could point to in in my Homeseer forum, so it would save you and @ebaauw on a step of explaining what is needed?

manup, 2018-09-21:
Good idea, I'll put up a wiki page with a mini howto on the required steps/information and maybe links to past examples. The process how it is done is often the same so the page should be very helpful.

ebaauw, 2018-09-21:
I second that!

Additionally, a FAQ or How-To Guide on pairing hard to pair devices would save us a lot if chatter on the issues.

wvuyk, 2018-09-21:
@ebaauw that would indeed be an nice extra for sure. In my forum there is a topic on this. The descriptions are aimed for usage in the plugin, But of course the first steps are describing the deCONZ part. Here is the [link to the topic](https://forums.homeseer.com/forum/lighting-primary-technology-plug-ins/lighting-primary-technology-discussion/jowihue-w-vuyk/106212-howto-add-deconz-compatible-devices-to-deconz)

Maybe this can be a start for input?

Is there a way I can help to add at least a few?

manup, 2018-09-21:
A first draft is now in the Wiki 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

> Additionally, a FAQ or How-To Guide on pairing hard to pair devices would save us a lot if chatter on the issues.

> Is there a way I can help to add at least a few?

We could start one Wiki page per device which are referenced from the [Suppored-Devices](../wiki/Supported-Devices) page. The page can provide all useful information. And also be references in GitHub issues and visa versa.

The new official documentation will also contain device specific documentation and REST-API descriptions. However the wiki pages are more easy to  edit in an open manner, so these can also be cross linked to the official documentation, which can integrate missing pieces if they are available earlier in the Wiki pages.

MattL0, 2018-10-04:
this would be a nice add https://www.amazon.com/dp/B00WTDV1TU/ref=psdc_7459508011_t2_B00HGVJRX2

simplextech, 2018-12-12:
Any updates for NYCE sensors?  A sensor joins right away but Phoscon doesn't see it.

simplextech, 2019-01-18:
Is there any additional information I can provide?

These sensors work with SmartThings, Hubitat Elevation and Vera Plus.  The SmartThings device handler is available on the NYCE site and upon request they will send you integration details.

manup, 2019-01-18:
Sorry for the delay it will be added in the next version for testing. Not sure about the handling of the motion signal yet as the sensor provides the Occupancy and IAS Zone clusters. I'll check the docs for some hints.

simplextech, 2019-01-20:
Update:  Joined the sensor last night and it joined/enrolled and is functioning.  Phoscon however does not see it as a sensor in the list but it is working and other apps (HomeSeer JowiHue plugin) can read the information fine through the REST API.

manup, 2019-01-20:
Cool, thanks for testing this.

>Phoscon however does not see it as a sensor in the list but it is working and other apps (HomeSeer JowiHue plugin) can read the information fine through the REST API.

It should be visible with 2.05.57.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_57

simplextech, 2019-01-20:
It is the latest dev branch.  I did a pull and compile for testing.  I didn't use a release as I don't think it's in a release just yet (I could/probably am wrong)

manup, 2019-01-20:
The Phoscon App is only part of the package not of this repo :)

simplextech, 2019-01-21:
Yup... I was just relaying the info :)

manup, 2019-01-21:
I've updated the wiki supported devices section.
Closing the issue for now.

closed 2019-01-21 by manup