dresden-elektronik/deconz-rest-plugin #549: Change presence sensors behaviour

sweebee, 2018-04-18:
Also already mention in #519 but want to create a new issue since I know what other bridges do.

Now when there is motion, deconz sends a presence false exactly after the timer expires. so when set to 60s  exactly 60s later it sends presence false. Even when there is still motion.  When the sensor sends motion again, deconz should reset the timer.

I know this is possible since Homey does the same. When I keep moving it won't reset until there is no motion for 60s and this works great with the Xiaomi/Aqara motion sensors.

But with deconz it is very annoying since I have to move again after it resets the presence to false.

manup, 2018-04-18:
>Even when there is still motion. When the sensor sends motion again, deconz should reset the timer.

The timer is already reset as soon as a signal motion signal is received, regardless if there was motion before or transitioning from no motion to motion.

I think the main issue is that the sensor does send these reports just very ~60 seconds (if it is running a while). Our default `config.duration: 60` is problematic here since sometimes the report is a bit off or even comes 2-3 minutes too late.

We should change the timeout for Xiaomi motion sensor to 120 or better 180 (240?) to decrease the change of missing a event. 

```C++
 if (ia->id() == 0x0000) // occupied state
                            {
                                if (updateType != NodeValue::UpdateInvalid)
                                {
                                    i->setZclValue(updateType, event.clusterId(), 0x0000, ia->numericValue());
                                }

                                const NodeValue &val = i->getZclValue(event.clusterId(), 0x0000);

                                ResourceItem *item = i->item(RStatePresence);

                                if (item)
                                {
                                    item->setValue(ia->numericValue().u8);
                                    i->updateStateTimestamp();
                                    i->setNeedSaveDatabase(true);
                                    Event e(RSensors, RStatePresence, i->id(), item);
                                    enqueueEvent(e);
                                    enqueueEvent(Event(RSensors, RStateLastUpdated, i->id()));

                                    // prepare to automatically set presence to false
                                    if (item->toBool())
                                    {
                                        if (val.clusterId == event.clusterId() && val.maxInterval > 0 &&
                                            updateType == NodeValue::UpdateByZclReport)
                                        {
                                            // prevent setting presence back to false, when report.maxInterval > config.duration
                                            i->durationDue = item->lastSet().addSecs(val.maxInterval);
                                        }
                                        else
                                        {
                                            ResourceItem *item2 = i->item(RConfigDuration);
                                            if (item2 && item2->toNumber() > 0)
                                            {
                                                i->durationDue = item->lastSet().addSecs(item2->toNumber());
                                            }
                                        }
                                    }
                                }
                                updateSensorEtag(&*i);

                            }
```

sweebee, 2018-04-18:
Don't know how it works in homey then, but I have set it also to 60 seconds and it works great. 

I think 120s is to long for a pir that triggers a light in a hall for instance. Why not default to 65 or 70 seconds?

manup, 2018-04-18:
Heres an sketch of the problem:

[image]

>I think 120s is to long for a pir that triggers a light in a hall for instance. Why not default to 65 or 70 seconds?

65 or 70 is fine if it works reliable all times, but I have seen sniffer logs were the sensor doesn't send a signal every 60 seconds for some reasons.

You can try it by setting the `config.duration` to 70 it might work well but it might also happen that a signal isn't send. I'm not aware of any configuration options on the sensor itself so we rely on the received presence signals.

> Don't know how it works in homey then, but I have set it also to 60 seconds and it works great.

https://github.com/TedTolboom/com.xiaomi-mi-zigbee

We should study how they do it then :)

sweebee, 2018-04-18:
Looks like they do it the same way:

```
		// Set and clear motion timeout
		clearTimeout(this.motionTimeout);
		this.motionTimeout = setTimeout(() => {
			this.log('manual alarm_motion reset');
			this.setCapabilityValue('alarm_motion', false);
		}, (this.getSetting('alarm_motion_reset_window') || 300) * 1000);
```

I've set alarm_motion_reset_window to 60.

manup, 2018-04-18:
Interesting, does it also work if it runs for some hours?

Note: When the sensor is powered/joined it does send the signal every 5 seconds only after few hours (?) the interval drops to 60 seconds.

sweebee, 2018-04-18:
Seems like it works fine when you've just added the sensor. When you have the sensor running for a day or more it will only trigger at 60s instead of quicker.

I've only tested in with homey when I just added the sensor. But seems after a while you have the same problem with homey https://github.com/TedTolboom/com.xiaomi-mi-zigbee/issues/59

TedTolboom, 2018-04-18:
@sweebee correct, the motion sensors (Xiaomi and Aqara) from my perspective have two device limitations (compared to other motion sensors)

1. the device only to sends the motion alarm trigger and not a cancellation; cancellation is only done from controller side
2. the device has 2 operation modes: normal user mode (retriggering once every 60 seconds) and test mode (retriggering once every 5 seconds, which is activated for 2 hours after inclusion or when the button is pressed), as explained in: https://forum.athom.com/discussion/comment/69062/#Comment_69062

I didn't find an option yet, to change the 60 seconds retriggering interval in the device / manufacturing specific parameters...

manup, 2018-04-19:
>Does the Heiman HS1MS work better @manup ?

I don't know, sadly haven't one yet. The Philips sensor and IKEA sensors should do since they send more frequently, Philips intervals can also be configured.

closed 2018-04-19 by sweebee