dresden-elektronik/deconz-rest-plugin #550: Firmware not showing after upgrade to 2.05.20

martikainen87, 2018-04-18:
Hi

Just did an upgrade to 2.05.20 and was going to upgrade the firmware on the deconz USB stick, but when I check the gateway info in phoscon app it aint showing.

[image]

Im running this in Docker with this code, maybe there was a change recently and I need to update my dockerfile? Last version I used was 2.05.05

> FROM ubuntu:xenial
> 
> ENV DEBIAN_FRONTEND noninteractive
> 
> RUN apt-get update && \
> apt-get install -y --no-install-recommends \
> ca-certificates \
> curl \
> libqt5core5a \
> libqt5network5 \
> libqt5widgets5 \
> libqt5gui5 \
> libqt5serialport5 \
> libqt5websockets5 \
> libqt5sql5 \
> sqlite3 && \
> apt-get clean
> 
> ENV DECONZ_VERSION 2.05.20
> 
> RUN curl -O https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-${DECONZ_VERSION}-qt5.deb && \
> dpkg -i deconz-${DECONZ_VERSION}-qt5.deb && \
> rm deconz-${DECONZ_VERSION}-qt5.deb
> 
> RUN chown root:root /usr/bin/deCONZ*
> RUN mkdir /root/otau
> RUN mkdir -p /run/udev/data/ && \
> echo "E:ID_VENDOR_ID=0403\nE:ID_MODEL_ID=6015" > /run/udev/data/c188:0
> 
> VOLUME ["/root/.local/share/dresden-elektronik/deCONZ"]
> EXPOSE 8080
> 
> CMD /usr/bin/deCONZ --auto-connect=1 -platform minimal --upnp=0 --http-port=8080 --dbg-error=1
> 
>

manup, 2018-04-18:
Nope no changes on that side, it may take a while (minutes) until firmware version is shown, this will be more fast in future versions.

martikainen87, 2018-04-18:
Alright, I'll wait untill tomorrow and see if it's showing, if not, do you know what firmware version i should use? I tried to find a list on this github but couldn't fine one.

manup, 2018-04-18:
Does it work otherwise: controlling lights, sensors, etc.?

Latest firmware is 0x261f0500 it's available in /usr/share/deCONZ/firmware.

Kane610, 2018-04-19:
I have the same issue, it shows after some restarts and can disappear again.

The correct firmware value is always printed in the logs

martikainen87, 2018-04-19:
Upgraded firmware to 0x261f0500, sorry to say but there is still no information about firmware in the phoscon app. It's no big deal for me but thought you would want to know. 

(This issue can be closed from my side even though there "problem" still exists)

manup, 2018-04-19:
The issue is on the todo list now, when everything else works it seems to be a display problem within the app, should be easy to fix.

gentzel, 2018-04-26:
This problem does appear to have been new with 2.05.20.  I see this consistently with 2.05.20 but never with 2.05.19.