dresden-elektronik/deconz-rest-plugin #551: How to complete setup steps without smart lights?

avraham, 2018-04-18:
Hi,

I'm trying to setup my conbee with this http://www.dresden-elektronik.de/pwa to be able to use it with Home Assistant. I believe I should get an API key once I complete the configuration.

The problem is I don't own smart lights, only xiamoi door sensors. So I'm stuck on the second screen where I'm supposed to turn on my lights. 

How can I get the API key to use conbee with Home Assistant?

manup, 2018-04-18:
Fear not: this will fixed soon in the Phoscon App.
Until then you can use following workaround:

instead of `/pwa/commissioning.html`
open `/pwa/index.html`

Then create a dummy group and add your sensors in *Menu > Devices > Sensors > Add sensor > Other*.

Note: with recent versions you may also use the local version `http://<raspberry-ip>/pwa`

avraham, 2018-04-18:
Thank you, that worked :)

manup, 2019-03-24:
The setup page now contains a short cut link to skip this step.

closed 2019-03-24 by manup