dresden-elektronik/deconz-rest-plugin #554: Download Ikea OTA files with python script

WinceWe, 2018-04-22:
Hey,

i´m trying to start the ikea-ota-download.py script with 

> python ikea-ota-download.py

to update my Ikea lights but i´m always getting 

> python: can't open file 'ikea-ota-download.py': [Errno 2] No such file or directory

What could be a reason for that? Normally it comes with the normal deCONZ installation or not?
I´m on python 2.7.13  and deCONZ 2.05.20 at the moment.

Thank you and greetings
Wince

Kane610, 2018-04-22:
Make sure you're in the same folder as the script

WinceWe, 2018-04-23:
I thought I was but it seems not :/ What´s the default directory of the file?

manup, 2018-04-23:
The script is not part of the package, you need to download it from here:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py

WinceWe, 2018-04-23:
Ah ok. Then I know what the problem is 😁 
Thanks alot!

jonathanweinberg, 2018-06-05:
@manup is there a way for me to contact you non-publicly?

closed 2019-03-30 by stale[bot]