dresden-elektronik/deconz-rest-plugin #555: Problems on Xiaomi devices

nerdosity, 2018-04-23:
Hello.

I have a lot of Xiaomi Devices, I used the Xiaomi Gateway until some weeks ago, but now I want to completely switch to Raspbee for everything since I am starting to mix Xiaomi (mainly for security and temperature measurement, their price are unbeatable) with other ZigBee products.
I already have a Osram RGBW device, and (apart from some lag) it works OK.

ATM I have:

- 1x Xiaomi gateway (the round one)
- 1x Xiaomi Mijiia smart button (round one)
- 1x Xiaomi Aqara window sensor (the squarish one)
- 2x Xiaomi Mijia window sensor (the roundish one)
- 3x Xiaomi temp/humidity sensor (the round one)

10x movement sensor (pir) are in the way.

I would like that everything just works, I need to trust every measurement, expecially battery and temperatures.

But i have to report some problems:

- door sensors get "stuck" if quickly closed-reopened. I played with them in the past and I don't really remember this behaviour
- battery status: that's too random. Someone told me in another issue that it get "up to one hour" to get refreshed. But that's a bit optimistic. Sometimes it got battery status immediately after adding, sometimes it got after multiple removing/readding, keeping them near and separating, sometimes it NEVER got battery status 
- temp/humidity seems too "static". I have more temp sensor in my home (DHT22 on ESP8266) and they are more quick to change. They draw a "curve" in home assistant, instead xiaomi's ones are usually on a flat line. 

Issue 1 and 3 could be Xiaomi's problems/behaviour, I can still test them on a real Xiaomi Gateway.
But on point 2 I can be absolutely sure: battery status, on my previous home assistant configuration (LAN gateway + door sensors) are reported IMMEDIATELY after closing/opening, or IMMEDIATELY after pressing the button (smart switch).

Tell me if I can help, I work as a web developer and I have experience in *nix servers. I don't mind losing some of my free time if it works toward having a complete Zigbee configuration and a reliable battery / status reporting from my sensors.

manup, 2018-04-23:
As referenced in the other post problem 2 should be fixed by now, there was an issue that Xiaomi seems to transport their manufacturer specific reports in two ways.

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ad615c67ac2a291dd4559e4232e77a4e45880567

The fix is available in the next deCONZ package, which should arrive in the next two days.

The temp/humidity sensors should now update at least hourly and faster when the related values changes significantly.

nerdosity, 2018-04-23:
Thank you.
Please let me know if I can help, I still have the Xiaomi Gateway, I am going to sell it only if Raspbee looks stable and reliable.

manup, 2018-04-23:
It would be interesting to know if the Xiaomi gateways allows any kind of sensor configuration.
Like can the motion sensor be configured to repeatedly trigger below 60 second intervals.

Currently we assume that the sensors work with default reporting intervals which is tricky in some cases especially for the motion sensor. 

Other projects seem to have the same challenge, would be great to study the Xiaomi gateway to shed more light on the topic.

https://github.com/TedTolboom/com.xiaomi-mi-zigbee/issues/34

nerdosity, 2018-04-26:
Hi,

motion/light sensor are not here yet.
In the meanwhile, I updated Raspbee firmware & updated deconz with this package:

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.24-qt5.deb
I can comment on third issue: now there is some "curve" in home assistant, it doesn't look like a straight line anymore. 

[image]

It's still a little "harsh", there are pretty big steps (60% -> 63%, one reading is almost one hour later) but it looks like much better. 
I would like to know if under quick changing it will report more changes, since I am going to use one of these thermo sensor for my heating thermostat.

I.e. this is a temperature reading from a DHT22 connected to a ESP8266:

[image]

nerdosity, 2018-05-07:
Hi,

half an hour ago I updated the deb with the latest beta one, now I just removed and reinstalled this sensor.

[image]
It stayed days without giving out battery readings.

This will be my final test, batteries are too much important on this kind of devices.
I will soon attach again the Xiaomi Smart Gateway, if it will behave better for batteries/temperature readings I will send the Raspbee back.

GoSpursGoNL, 2018-05-09:
At least you are able to reinstall your sensors, [I'm not](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/204#issuecomment-387475632). If you have a tip, I'd gladly hear it.

abmantis, 2018-06-22:
For me, all xiaomi devices connected to deconz always show 100% battery. Devices connected to the xiaomi gateway show much less.

closed 2018-11-23 by manup