dresden-elektronik/deconz-rest-plugin #556: Touchlink reset in phoscon

sweebee, 2018-04-25:
In one of the latest versions I get redirected directly to the phoscon app. But some times I used the old version to reset lights with touchlink. This is not possible with phoscon? I'm using the direct api calls now but that is not very convenient.

I can't acces the old interface, it keeps me redirecting to phoscon.

manup, 2018-04-25:
Hi we are still not sure to add Touchlink in the Phoscon App but it has some implications in usability which are quite difficult.

> I can't acces the old interface, it keeps me redirecting to phoscon.

In *Phoscon App > Menu > Help* there is a Link which directs to the old WebApp.

FezVrasta, 2018-05-12:
Hi @manup, I'm trying to join a Tradfri bulb but it's not being discovered. I successfully joined another one of the same model.

I tired to use the touchlink because with the Hue bridge this usually worked, and in fact I can make it blink once the touchlink has discovered it, but still I can't manage to make it join the network, do you have suggestions? Thanks

closed 2018-05-29 by sweebee