dresden-elektronik/deconz-rest-plugin #557: IKEA trådlös dimmer buttonevents

PremiumUsername, 2018-04-26:
When I try to use the IKEA dimmer (the rotating one) it produces buttonevents but they do not correspond to the direction of the spinning. Clockwise and counterclockwise both produce a log such as this:

"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2018-04-26T12:00:00"},"t":"event"}"
4/26/2018, 1:59:58 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]
"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2018-04-26T12:00:00"},"t":"event"}"
4/26/2018, 2:00:27 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]
"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2018-04-26T12:00:29"},"t":"event"}"
4/26/2018, 2:00:27 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]
"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2018-04-26T12:00:29"},"t":"event"}"
4/26/2018, 2:00:27 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]
"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2018-04-26T12:00:29"},"t":"event"}"
4/26/2018, 2:00:28 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]
"{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2018-04-26T12:00:30"},"t":"event"}"
4/26/2018, 2:00:28 PMnode: 3343d3e3.4f8e4c
msg.payload : string[115]

This is true of both the old firmware for the dimmer and the new one.

Kane610, 2018-04-26:
2002 and 3002 are different directions so it does work

PremiumUsername, 2018-04-26:
I know those are different directions. I was only spinning one way when that log was made.

mariusmotea, 2018-05-01:
Convert the hue rules from [here](https://gist.github.com/mariusmotea/eb9e778279b424a14d4f9faf83108bd7) to your project and it will work as expected.

FezVrasta, 2018-06-30:
I'm having the same problem, I think there's too much sensibility and so the dimmer fires events even for micro rotations in the opposite direction. The dimmer triggers events even if you are just moving it slowly at mid air without any rotation.

closed 2019-03-30 by stale[bot]