dresden-elektronik/deconz-rest-plugin #558: Firmware update selects wrong device

ebaauw, 2018-04-26:
I've got a Raspberry with both a RaspBee (for deCONZ) and a ConBee (for ZShark) connected.  When doing a firmware update from the web app, deCONZ updates the ConBee firmware instead of the RaspBee firmware.

Workaround is simple: disconnect the ConBee before doing the update.

manup, 2018-04-27:
Currently the firmware update code is pretty basic and doesn't handle established connections, in the above case it automatically should detect that RaspBee is the actual current deCONZ device.

I'll put it on the list.

ebaauw, 2018-11-23:
While I appreciate the effort to clean stale issues, I’d hate to see this fall off the “list”.

manup, 2018-11-23:
Yeah a bit of stale storm.. I ll need to cherry pick the ones to be active again, feel free to remove the wontfix label

closed 2019-03-30 by stale[bot]