dresden-elektronik/deconz-rest-plugin #560: Cannot pair Xiaomi Button

SlayDictator, 2018-04-27:
I'm unable to pair a Xiaomi button using Phoscon. 
I know that I have to add it as a sensor and I know also that it won't show up in the user interface after pairing.
However, I'm unable to complete the pairing dialog, it just won't pair. I'm using a RaspBee with a Raspberry Pi 3.

Version 2.05.23 / 24.4.2018
Firmware 261F0500

manup, 2018-04-27:
Does it appear in the deCONZ gui?

manup, 2018-04-27:
Yes currently not all Xiaomi devices are shown in Phoscon App, but we plan to support all of them.

So the second button isn't visible in REST-API? I'm asking because if the sensor is already registered the pairing dialog might not complete (albeit it should, I need to check it).

SlayDictator, 2018-04-27:
Since I'm using the REST-API it's no problem for me that the buttons aren't listed in Phoscon.

Currently I'm trying to pair two more buttons and both of them do not show up in the REST-API. The one that was paired before (with the software that was used before Phoscon) is still working fine.
Since I can't pair both of the new buttons I assume that the hardware is ok.

SlayDictator, 2018-04-27:
I think I have to correct this:
I just tried to pair an IKEA Tradfri Remote Control just like described in the pairing dialog - also impossible. Seems that I can't pair anything?!

manup, 2018-04-28:
Do you have by any chance, the sensors/switches view open in multiple browsers or tabs?
Current implementation is a bit stubborn in this case.

SlayDictator, 2018-04-28:
Just checked: No open browser tabs.
I also stopped HomeAssistant and checked the traffic on the port using tcpdump. Opened a new tab in a different browser (Chrome instead of Safari) and tried to pair a button and the Tradfri Remote Control again - no luck.

SlayDictator, 2018-04-28:
Got it. I tried several restarts to solve the problem, that didn't work. Tried now to completely shut down the system and removed the power supply. After booting up again I'm now able to pair devices again.
The buttons still do not show up in Phoscon but they are working via the REST-API.

closed 2018-04-28 by SlayDictator