dresden-elektronik/deconz-rest-plugin #562: deCONZ for arm7l

alpintrekker, 2018-04-27:
Hi Guys,

I'm pretty sure, that this question is a bit offtopic, however, you may know the answer:
who do I need to contact at Dresden Electronic to get deCONZ binaries cross compiled for arm7l architecture? RaspBerry is armhf...

Regards,
András

alpintrekker, 2018-06-14:
@manup, maybe you could give me a hint about this question?

manup, 2018-06-14:
Hello, sorry currently we only support the default Raspbian for ARM and there are no short term plans to support  and maintain armhf. This may change in future if the demand rises.

alpintrekker, 2018-07-04:
@manup, are you open to the proposed cooperation?

> Hi, bad news. It is actually BeagleBoneBlack's architecture. Up to my understanding, it would be just a different cross-compilation.
However, I would happy to contribute with cross-compiling, even in the official way, under terms of an NDA

manup, 2018-07-05:
>@manup, are you open to the proposed cooperation?

Hi, I'm afraid currently this isn't an option for us. However we are currently working more and more build automations in virtual environments, are there QEmu images for the BBB system?

alpintrekker, 2018-07-19:
Actually, there are some, that claim, they are, but never tested.

closed 2019-03-30 by stale[bot]

johanneswilm, 2019-12-05:
@manup This is actually quite an important point still. I was quite surprised when I discovered today that for some reason you guys are blocking this and so therefore my new conbee 2 is basically useless.