dresden-elektronik/deconz-rest-plugin #564: Cannot achieve a light flash

ooii, 2018-04-28:
Hi,

I'm trying to make a light flash.
I sent a `put` request to `/lights/10/state` with body `'{"body":{"alert":"select"}}'`. My request receives a `200` http code but the light does not flash.
What am I doing wrong?

ebaauw, 2018-04-28:
The body should be `{“alert”: “select”}`.  Don’t be fooled by the 200 OK response; deCONZ does lousy error handling of invalid attributes in the body.  It does return the set attributes in the response body.

ooii, 2018-04-29:
Thanks @ebaauw.
Actually, I'm trying to achieve the following scenario:
When a dimmer button is pressed, (1) flash light immediately, then (2) flash light 1 min later. For that, I created a rule that (1) triggers the light flash and (2) enables a schedule that triggers flash light 1 min later.
The first flash is ok but the second one never happens.
Can't find where is my error.
Here are the rule and the schedule, if you could tell me please. Thanks.

PS: When I long press the button, the first flash is triggered. But the rule always appears as never being triggered (`lasttriggered` is `none` and `timestriggered` is `0`).
PPS: Is it normal that if I send a `put` request to `/schedules/2` with body `'{"status":"enabled"}'`, its status remains to `"disabled"` (and it does not fire the flash)?

```
{
    "actions": [
      {
        "address": "/lights/1/state",
        "body": {
          "alert": "select"
        },
        "method": "PUT"
      },
      {
        "address": "/schedules/2",
        "body": {
          "status": "enabled"
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/5/state/buttonevent",
        "operator": "eq",
        "value": "3003"
      },
      {
        "address": "/sensors/5/state/lastupdated",
        "operator": "dx"
      }
    ],
    "created": "2018-04-27T19:51:06",
    "etag": "8c112967619e744b94f8b593528465c8",
    "lasttriggered": "none",
    "name": "Test pour brossage dents",
    "owner": "myName",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  }
}
```

```
{
    "activation": "start",
    "autodelete": false,
    "command": {
      "address": "/api/<username>/lights/1/state",
      "body": {
        "alert": "select"
      },
      "method": "PUT"
    },
    "created": "2018-04-27T19:48:38",
    "description": null,
    "etag": "9952f38df3c1bd025c65357466c94b32",
    "localtime": "PT00:01:00",
    "name": "Brossage De Dents",
    "starttime": "2018-04-27T19:48:59",
    "status": "disabled",
    "time": "PT00:01:00"
  }
```

ooii, 2018-04-29:
OK, thanks. Did it this way.

closed 2018-04-29 by ooii