dresden-elektronik/deconz-rest-plugin #565: Siren Heiman HS2WD-E

jsponz, 2018-04-29:
Hi,

I have just bought one of this devices. 

I can add it via Phoscon App. It is detected as a light. 

The problem is that I can't switch it on or switch it off. From the web panel, there is no reaction. However if I edit a group and click on the device, it sounds (identification phase). 

Any help!

Thanks!

ebaauw, 2018-04-29:
Check the light resource: it doesn’t contain `state.on`, only `state.alert`.  This is by design: I don’t want the Siren to react when switching all lights on.  You can use values `”select”` and `”lselect”` to make the siren scream and flash; `”blink”` to make it flash silently, and `”none”` to make it stop.  I suppose PWA doesn’t yet support the siren.

jsponz, 2018-04-30:
Thanks! It works!

PWA shows the siren, but it is not possible to control it.

closed 2019-03-30 by stale[bot]

martinschm, 2020-05-09:
Hi,
I bought the same siren. I could add it and can now see it in the Phoscon App as a "warning device". But no way of controlling flashing or sound via the web interface nor in my smart home software (iobroker).

Any chance of getting this properly integrated?

kilasat, 2020-07-02:
> Hi,
> I bought the same siren. I could add it and can now see it in the Phoscon App as a "warning device". But no way of controlling flashing or sound via the web interface nor in my smart home software (iobroker).
> 
> Any chance of getting this properly integrated?

Did you try this?

Check the light resource: it doesn’t contain state.on, only state.alert. This is by design: I don’t want the Siren to react when switching all lights on. You can use values ”select” and ”lselect” to make the siren scream and flash; ”blink” to make it flash silently, and ”none” to make it stop. I suppose PWA doesn’t yet support the siren.