dresden-elektronik/deconz-rest-plugin #566: TRÅDFRI Motion sensor not connected anymore

jsponz, 2018-04-29:
After the last DECONZ update, TRÅDFRI Motion sensor is not detected anymore. 

Any hint?

Thanks!

fabianhenniges, 2018-04-29:
My TRÅDFRI remotes are gone too. All 9. Updated from 21 to 24.

jsponz, 2018-04-30:
Nothing :-(

mariusmotea, 2018-05-01:
Same problem for me after i upgrade from Raspberry 2 to Raspberry 3+. 2 sensors were automatically joined back to the new deconz installation , but 3rd sensors failed every time. I replaced the batteries and i test also the sensor with Tradfri Gateway, so the sensor seams ok.

djwlindenaar, 2018-05-02:
I have the same issue. Stable network for a couple of months, now after update to version 2.5.24, some tradfri sensors started dropping off the network. Not all of them. 2 out of 6 RHT sensors and 1 out of 2 motion sensors so far. Re-pairing does not seem to help.

Two of the dropped off sensors are meshing through a tradfri light. That light shows the red dot sometimes in the UI. Could be related? 

Let me know how I can help solve this regression!

Daniel

manup, 2018-05-02:
Is the firmware of RaspBee the latest: 0x261f0500?

Not sure what happens here, but before reset anything please always try to just pull out the battery press buttons a few times and reinsert battery.

If possible try to force the devices to connect though the RaspBee directly by turning power of the IKEA light off before the "battery-toggle".

djwlindenaar, 2018-05-02:
Yes, it's the latest firmware. 

I got things working again by powercycling the IKEA light and then rejoined the sensors through the light (as it was before updating deconz). 

This worked out fine and things are now communicating again. I hope this was just related to the update event and not a stability issue. We'll see..

djwlindenaar, 2018-05-03:
Hmmm that sound like a completely different issue to me. The motion sensors I have both created a group and a sensor in the REST API after joining the network.

I guess my issue was more in the light which seemed to have an issue than in the sensors.

Daniell

mariusmotea, 2018-05-26:
Is there a way how i can debug this issue? I still cannot pair this sensor after multiple retries.

alexvandervegt, 2018-05-27:
I have the same issue

am3rig0, 2018-06-24:
Just wondered if anyone had any luck with getting TRÅDFRI Motion sensor to work on 2.05.30? Had a go at reincluding my standalone unit (with no lights attached) that worked fine before 2.05.25.  

No matter what I try (replacing / removing battery, multiple resets, connecting conbee using usb extension), still no luck connecting this motion sensor in Phoscon.

mariusmotea, 2018-06-24:
I did not find any workaround for this issue.

manup, 2018-06-25:
I've test two scenarios now (deCONZ 2.05.31, Firmware 0x261f0500)

1) Reset motion sensor without deleting it, to check if it will be detected again when the NWK address changes.
2) Delete motion sensor from the Phoscon App so it will be marked *deleted* in the database and when add it again via Phoscon App.

Both tests worked, so currently I can't spot a general bug in deCONZ on the matter.

Can you please retry to add the sensor but with all lights (and other mains powered) devices powered off, that is to ensure the sensor will join via gateway. This should minimize eventual problem on in between nodes.

am3rig0, 2018-06-25:
@manup thanks for looking into this. 

My main system is a RPI running hassio image, so I am relying on marthoc/hassio-addon-deconz-armhf docker image for latest version of deCONZ.

I did however spin up an ubuntu 18.04 system in virtualbox and can confirm that deCONZ 2.05.31 ( on Firmware 0x261f0500) actually detects the sensor just fine (via USB passthrough).

Once @marthoc had a chance to update his image, i'll check this on my hassio box, fingers crossed!

mariusmotea, 2018-06-25:
I try 30 minutes today but with no luck. I have Deconz 2.05.31 / firmware 26210500. Not sure if this helps but while i try to pair the sensor the green light has a loop (10 seconds on, 1 second off). Because of no luck i try to keep the link button pressed for about 10 seconds and i believe the raspbee device was reseted because i saw ~3 fast red flashes and lights control stop working.

am3rig0, 2018-06-25:
@mariusmotea perhaps try firmware 0x261f0500 - this seems to work for both myself and @manup

manup, 2018-06-27:
To downgrade the firmware please do:

**RaspBee**
1) close deCONZ
2) `$ sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`

**ConBee**
1) close deCONZ
2) `$ sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`

mariusmotea, 2018-06-29:
Unfortunately it did not pair also with firmware 0x261f0500. I will dig more, maybe i will find our what is going on. There is no way to start deconz in debug mode?

manup, 2018-07-01:
>There is no way to start deconz in debug mode?

You can enable (noisy) logging by starting deCONZ on the command line like:

```
$ deCONZ --http-port=80 --dbg-error=1 --dbg-info=2 --dbg-aps=2 --dbg-zcl=2
```

There should be hopefully something to see when the sensor joins.

mariusmotea, 2018-07-01:
I believe the issue is with deconz configuration. I start deconz with above command and it start with a fresh configuration (i guess it use my user env, not root where current configuration is stored). On a light search it detect all my lights instant, however no switches or sensors were detected. I try to pair the sensor that did not work in the past and now there was absolutely no problem to pair it. Now i have two configurations, one were i have just this sensor and another one with others, excepting this.

manup, 2018-07-03:
That might indeed be possible, the zll.db file could contain references to the old sensor, usually marked as deleted. deCONZ normally should take care of it and bring back the sensor while searching (as tested in the above two scenarios). To find out the underlying problem  you need to use the former configuration, and maybe track where the setup process stops.

manup, 2018-11-23:
Should work with recent deCONZ versions, there were cleanups in regard to database handling.

closed 2018-11-23 by manup