dresden-elektronik/deconz-rest-plugin #569: Get 'added' signals for all types in deconz over web socket

Kane610, 2018-04-30:
Hi @manup !

I'd like to support live additions of new devices, groups, etc. in home assistant. But there seems to be many cases that aren't supported today. Do you think this could be possible to support?

manup, 2018-05-02:
Yes I also think this is very useful, currently sensors do emit this added event, not sure about lights. I put it on the todo list.

Kane610, 2018-05-02:
I'm about to get initial support for sensors integrated to HASS.
Just need lights, groups and scenes now 👍🏻

Kane610, 2018-05-02:
Oh! Also: I noticed that sensors with battery doesn't report battery on added, IA that device dependent or is that something that gets reported later?

manup, 2018-05-02:
In case of the Xiaomi sensors I think either they don't send this report early on or also likely it it received before the sensor resource is initially created, hence the delay of up to one hour until next report is received.

Kane610, 2018-09-04:
@manup any eta on this? From an outside perspective it shouldn't be that extensive to add lights and groups to do this signalling.

manup, 2018-09-04:
Hmm I thought lights already announce this via websocket &mdash; state/reachable?
I need to check it :)

manup, 2018-09-04:
Also as a side note, deCONZ 2.05.37 will load all lights directly after start not waiting for them to be discovered.

Kane610, 2018-09-04:
@manup thats a good improvement for deconz with loading all lights directly!

I haven't seen any notification for new lights over websocket

closed 2019-01-09 by stale[bot]