dresden-elektronik/deconz-rest-plugin #571: Meshing not working?

sweebee, 2018-04-30:
I have a Tradfri LED dimmer for 2 months or so. But since today it stopped working if its more that about 1-2 meters away from the gateway. At the default place its around 4 meters away from the gateway. There is 1 other tradfri light 2 meters away from the dimmer and that one works fine.

Could the leddriver be broken or does meshing not work? The other tradfri light is almost in line of sight and works fine.

I also have 7 lights In the basement. 3 of them are Hue lights and 4 of them tradfri. The one closest to the gateway works fine (around 2 meter straight under the floor). But the others won't turn off/on sometimes. they are 3/4 meters from the other light and around 5/6 meters away from the gateway.

mariusmotea, 2018-05-01:
Interferences with your local wifi network. Try changing the channels following this [guide](https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence)

sweebee, 2018-05-02:
Changed my AP's to channel 1 and 6 and for the third I disabled 2.4 ghz. Can I change the zigbee channel without any problems? The lights will change automatically?

closed 2018-05-05 by sweebee