dresden-elektronik/deconz-rest-plugin #573: Dresden windows release site

theendisnye, 2018-05-01:
Is anyone else having problems getting to the Dresden windows release site https://www.dresden-elektronik.de/deconz/win/ or has it moved?

Steve

saurabh984, 2018-05-01:
Having problems connecting to all of https://www.dresden-elektronik.de

manup, 2018-05-01:
Looks like the server has crashed, today is holiday, it should be online again tomorrow morning.

manup, 2018-05-02:
Short note: the problem still remains, we are working on it..

ghdbwise, 2018-05-02:
Is it possible to find the .deb-files on another server?

ImEmJay, 2018-05-02:
Their server is back online. 

I also searched for a mirror for the last days but didn't find any other source.

alexvandervegt, 2018-05-02:
@manup nice work, can you also post the .deb file for version 25?

manup, 2018-05-02:
> @manup nice work, can you also post the .deb file for version 25?

Yes will be up till tomorrow :)

Fl0ca, 2018-05-04:
Is it possible to update the Ubuntu Releases to 25, too?

Thanks :)

manup, 2018-05-04:
Yes should arrive in a few hours.

manup, 2018-05-05:
>Is it possible to update the Ubuntu Releases to 25, too?

Took a while, it's online now :)

jonathanweinberg, 2018-06-05:
This should probably be closed?

closed 2018-06-05 by theendisnye