dresden-elektronik/deconz-rest-plugin #574: tradfri light is missing - still responding to group commands

ErnstEeldert, 2018-05-03:
I am missing one of my IKEA tradfri lights in the deconz web interface. I can't see it in the group overview, nor the available lights overview. The light does respond to group commands, so it seems to still be connected to the network?

I switched the lights of the group off and on, no change.

I upgraded (.25) and restarted deconz, no change.

How can I get my light back?

manup, 2018-05-03:
Can you read the basic cluster attributes in deCONZ GUI, maybe something like modelid or vendor is missing and the light is thereby not shown in the web interface.

ErnstEeldert, 2018-05-07:
Looks like the only clusters left on this particular light are On/Off and Level Control, hence it's still playing nice with the group. This is what the node info looks like:

[image]

ErnstEeldert, 2018-05-14:
Any ideas around this? Would like to reset the lamp to get this fixed, so if you want any more diagnostic info in its current state, let me know.

sweebee, 2018-05-20:
I've got the same issue with 4 (tradfri) lights. Tried to reset them but deconz will not find the lights anymore.

sweebee, 2018-05-20:
I also bought a new tradfri light and I could add it directly. But with the existing ones that disappeared from deconz, I can't do anything. Tried like resetting it 20 times, nothing works. I can add these lights without any problem to the ikea gateway.

manup, 2018-05-22:
By disappear you mean in the deCONZ GUI window or WebApp/Phoscon App? In the deCONZ GUI lights might be a bit out of the visible window, you may zoom out with the mouse scroll wheel to find them.

ErnstEeldert, 2018-05-23:
So decided to get rid of this particular light, but turns out I can't 🙈 

Tried the reset procedure using the 6xpower cycle procedure. It blinks shortly after, but nothing happens when looking for new lights, so it refuses to re-pair.

In deconz, I can still see the light in the node list, and it still just has the same clusters as shown in the screenshot above. It also still responds to On/Off commands for example.

I tried to do a touchlink reset, it shows up after a scan, but the reset command doesn't seem to affect it. Tried multiple times.

So, how do I reset this light, besides returning it to IKEA for a replacement?

ebaauw, 2018-05-24:
> It also still responds to On/Off commands for example.

Then it hasn’t been reset.

> I tried pairing it to the Tradfri gateway (which I normally don't use), and it worked just fine.

Once you pair it with the Trådfri gateway, it should no longer be controlled by deCONZ.

> Deleted the node from the GUI, did a reset, nothing seems to help

The light is still cached in the neighbour tables from other lights, forming the mesh network.  Wait a couple of days, or power-cycle all lights.  After that, the node shouldn’t re-appear anymore after deleting it from the GUI.

ErnstEeldert, 2018-06-09:
I waited for a few days to no avail, light would not repair/show up on deconz again. Also not sure as to why it wouldn't reset fully, as I was pretty sure I was doing the right procedure (seems to work with other lights), but the response on the reset was a short blink, not the pulse I was expecting.

In any case, I gave up and replaced it by a brand new light.

closed 2018-06-09 by ErnstEeldert

ebaauw, 2018-06-09:
If you have a Hue dimmer switch, you might try and reset the light by holding the switch next to it (< 15 cm) and pressing and holding On and Off buttons for 10 seconds.  Maybe you need to power cycle the light (once) just before that.

Alternatively, you might try to reset the light using deCONZ' touchlink (_Menu|Settings|Scan for devices_ in the old web app).  Again, hold the light next to the RaspBee or ConBee, and you might need to power cycle the light just before that.  If you cannot move the light, use a USB extension cable to move the ConBee, or run your Raspberry from a USB power bank and move the RapsBee.  If the light was reset successfully, it should report a different network id on the next scan.

To pair (stubborn) devices to deCONZ, power off all ZigBee routers (yellow nodes in the deCONZ GUI), move the light in range (couple of meters) of the RaspBee/ConBee.  This ensures the device connects to the RaspBee/ConBee directly, instead of over the mesh network.  Open the network through the webapp (_Menu|Settings|Open Network_) or Phoscon app (search for new lights), and power cycle the device.  It should appear in the deCONZ GUI.  Then, while the network is still open, manually read the _Basic_ cluster attributes in the _Cluster Info_ panel to make sure deCONZ identifies the node correctly.  For battery-powered devices, press the (reset) button, move it, wave in front of it, or something, to ensure it remains awake when reading the _Basic_ cluster.

Kendobendo, 2018-06-12:
I just thought I'd throw my two cents into the pot. I have noticed that since a while back a couple of my lights from IKEA just suddenly one day decided to not show up in deCONZ anymore. I didn't bother trying to solve this a a few weeks as I had other things to work on.

Fast forward a bit and I did some digging. The two lights that are currently gone as still in the SQLLite databas that deCONZ is using. And I am unable to remove the lights as they dont show up in the UI. I'm runing deCONZ on Hassio in a docker so I'm not sure if I actually can access the REST API. 

I've tried to reset the bulbs. I even did the on/off six times. The light blinks once after the reset to confirm the reset. I am unable to control the lights. Trying to trigger on off on the group they were previously in before does not work as well. I am also unable to add them back by touchlink. My best guess would be that deCONZ refuses to add the lights since they are still listed in the database. I tried removing the faulty lights from the database and I did clean up sessions.default as well. But it made no change.

I am able to pair the light with my old IKEA Gateway and I tried pairing it up with my friends Conbee and that works fine too. So this is all weird. I guess the last resort would be to simply do a gateway reset and see if that works. 

I'm currently runing deCONZ 2.05.28 and firmware 26210500. I guess I could try to have all the lights updated by the IKEA Gateway. I'm not sure if there's a newer firmware version available. I haven't checked. I'm currently running:
4x TRADFRI bulb E27 CWS opal 600lm v1.3.002
3x TRADFRI bulb E27 WS opal 980lm v1.2.217

Any ideas what else I could try before doing a gateway reset?

lauer, 2019-08-28:
> If you have a Hue dimmer switch, you might try and reset the light by holding the switch next to it (< 15 cm) and pressing and holding On and Off buttons for 10 seconds. Maybe you need to power cycle the light (once) just before that.

This solved the missing connection for me. Didn't have to delete the anything first. It made the deCONZ webinterface find the lights again.
When holding the ON and OFF buttons for 10 seconds, the little red light on the Hue Dimmer will flash and after a second or two, the bulb will also flash. Then release the buttons. 
Toggle the light afterways and just wait for the reconnections, which took ~10 seconds.