dresden-elektronik/deconz-rest-plugin #578: Join networks?

FezVrasta, 2018-05-06:
Hi, sorry if I'm using the issue tracker to ask a question but I'm not sure where else to ask it.

I have a Hue Bridge 2.0 and a RaspBee, everything is linked to Home Assistant.

I use the Hue Bridge to handle the network that controls the bulbs, this because my Philips TV with Hue integration doesn't support 3rd party bridges.
The RaspBee is used to control some IKEA Tradfri remotes, that thanks to Home Assistant can control my bulbs.

I have a mix of Hue and IKEA bulbs in my home, the IKEA ones have a bug that makes them flicker once in a while if paired to the Hue Bridge.

I'd like to pair all my IKEA bulbs to RaspBee, but still have them show up in the Hue app.

I think I need a way to tell DECONZ to join the Hue bridge somehow, is this something possible to do?

Thanks

ebaauw, 2018-05-06:
> I think I need a way to tell DECONZ to join the Hue bridge somehow, is this something possible to do?

Short answer: no.

Long answer: In the past, we used to be able to join a RaspBee as a router to the ZigBee network created by the Hue bridge, see #21.  For some reason, specifying the (once secret) ZLL Link Key in the deCONZ GUI doesn't stick, and this no longer seems to work.  Even if it would, both gateways would be controlling (and polling!) the lights and, if you're not careful, would compete for the graces of the sensors and switches, who generally can report to one gateway only.

You might try and connect the Hue app to deCONZ instead of to the Hue bridge.  As long as you have no exotic devices connected, this seems to work.

FezVrasta, 2018-05-09:
Thanks for the info. I can't use the deCONZ directly because as said it doesn't work with my TV, also, it seems like that my Osram Power Outlet makes the Hue app crash :-(

FezVrasta, 2018-05-09:
Anyway, I found this project that promises to expose any DeCONZ light as if they were handled by a Hue Bridge, I'll give it a try and report back.

https://github.com/mariusmotea/diyHue

yene, 2018-08-04:
Sorry to bring this old issue back alive. 
I would like to join an existing IKEA network and read the values of end devices. Later do the same with Xiaomi. Is this possible?

FezVrasta, 2018-08-04:
The situation didn't change, AFAIK.

Just for completeness, I tried diyHue, but, I'm not sure if because of my hardware (RPi3 B) or because of the sofware, the lights lagged a lot.

closed 2018-12-10 by stale[bot]