dresden-elektronik/deconz-rest-plugin #580: Hue Motion sensors are recognized at one

kappes, 2018-05-08:
I'm running Produkt RaspBee Version 2.05.23 / 25.4.2018 Firmware 261F0500 on RasPi3/Raspbian Stretch. 

I've got two Hue Motion sensors, when I try to connect them to the bridge both seem to connect just fine, but only the first is recognized by the bridge. I tried multiple reinstalls and repairs, only the first sensor that gets connected is greeted on pair (connected lights are blinking), but the system is updating values for one or the other sensor (I guess, it's always the last that I connected, but I cannot verify that). 

In my last attempt, the bridge even renamed an existing sensor ("Bewegungsmelder" -> "Bewegungsmelder (2)") and stopped displaying the existing sensor. 

It seems like the bridge cannot separate the sensors.

manup, 2018-05-09:
>It seems like the bridge cannot separate the sensors.

Indeed but that's odd, it would mean that both sensors have the same mac address, which is very unlikely.

Do you see both sensors as nodes in deCONZ?

kappes, 2018-05-09:
one disappears completely when the other is connected.

kappes, 2018-05-11:
I deleted the nodes in deconz. after an reconnect, I was able to see both sensors again. I still cannot connect both at the same time, but at least they show up in the mesh now.

closed 2018-05-11 by kappes

kappes, 2018-05-15:
both sensors appear to have the same MAC, this is strange 
[image]

ebaauw, 2018-05-15:
I find it very hard to believe that Philips would ship two motion sensors with the same ZigBee MAC address.  More likely, there's a stale reference somewhere in your ZigBee network.

If you still have a Hue bridge, can you reset the sensors, pair them with the Hue bridge, and check the `uniqueid` values for the sensor resources.  They should show different MAC addresses.

If they do, you might try to:
- Reset the sensors (pressing the reset button/hole for 10 seconds, until the LED starts blinking);
- Delete the corresponding node(s) from the deCONZ GUI;
- Delete the corresponding REST resource(s) for these sensors.  Make sure to delete the ZHAPresence, ZHALightLevel, and ZHATemperature resources.
- Stop deCONZ;
- Delete the sensors records from the deCONZ database (they're now marked 'deleted');
- Reset the RaspBee (power cycle the Raspberry or use `sudo GCFFlasher_internal -r`);
- Power cycle all your lights (and other ZigBee routers);
- Start deCONZ and re-pair the sensors.

manup, 2018-05-15:
I wonder how the above screenshot is possible, deCONZ doesn't allow multiple nodes with same mac address (in fact it's the only way to distinguish nodes) changing NWK addresses are normal and to be expected after each reset of a device. 

Can you make a screenshot from the both nodes in the deCONZ GUI?

kappes, 2018-05-16:
I deleted the nodes in the UI, checked the REST API, deleted the deleted marked entries from the sqlite3 db and restarted everything. The duplicated MAC entries are gone now and I was able to pair both sensors successfully (all my lights blinked to acknowledge the sensors), but only one is showing up in Phoscon. 
At the moment "Bewegungsmelder Flur" is not really updating. I give it a little time and see if it actually works. I have no idea why the other sensor won't show up in the interfaces.

I attached screenshots of the deCONZ GUI, as well as the current sensors table below. If there is any more meaningful output I can provide, please let me know. Also, if I should create a new issue for this?

**deCONZ GUI**
[image]
[image]

**Current sensors table:**
```sqlite3
10|Dimmer Schlafzimmer|ZHASwitch|RWL021|Philips|00:17:88:01:10:5f:69:77-02-fc00|5.45.1.17846|{"buttonevent":1002,"lastupdated":"2018-05-11T19:36:38"}|{"battery":100,"group":"7087","on":true,"reachable":true}|{"d":2096,"ep":2,"in":[0,1,64512],"out":[6,8,5],"p":49246}|normal|1
9|Dimmer Büro|ZHASwitch|RWL021|Philips|00:17:88:01:10:5f:6c:84-02-fc00|5.45.1.17846|{"buttonevent":4002,"lastupdated":"2018-05-11T23:02:25"}|{"battery":100,"group":"18381","on":true,"reachable":true}|{"d":2096,"ep":2,"in":[0,1,64512],"out":[6,8,5],"p":49246}|normal|1
1|Daylight|Daylight|PHDL00|Philips|00:21:2e:ff:ff:02:2e:f0-01|1.0|{"daylight":true,"lastupdated":"2018-05-16T04:07:45","status":160}|{"configured":true,"lat":"47.070714","long":"15.439504","on":true,"sunriseoffset":30,"sunsetoffset":-30}||normal|1
3|Power 3|ZHAPower|Plug 01|OSRAM|84:18:26:00:00:0b:58:51-03-0b04|V1.05.09|{"lastupdated":"2018-05-16T10:26:06","power":0}|{"on":true,"reachable":true}|{"d":16,"ep":3,"in":[2820],"p":49246}|normal|1
8|Power 8|ZHAPower|Plug 01|OSRAM|84:18:26:00:00:0a:48:11-03-0b04|V1.05.09|{"lastupdated":"2018-05-16T10:23:40","power":0}|{"on":true,"reachable":true}|{"d":16,"ep":3,"in":[2820],"p":49246}|normal|1
6|Bewegungsmelder Flur|ZHAPresence|SML001|Philips|00:17:88:01:02:03:27:b8-02-0406|6.1.0.18912|{"lastupdated":null,"presence":false}|{"alert":"none","battery":100,"delay":0,"ledindication":false,"on":true,"pending":0,"reachable":true,"sensitivity":0,"sensitivitymax":2,"usertest":false}|{"d":263,"ep":2,"in":[0,1,1030],"p":260}|normal|1
7|Bewegungsmelder Flur|ZHATemperature|SML001|Philips|00:17:88:01:02:03:27:b8-02-0402|6.1.0.18912|{"lastupdated":null,"temperature":null}|{"alert":"none","battery":100,"ledindication":false,"offset":null,"on":true,"pending":0,"reachable":true,"usertest":false}|{"d":263,"ep":2,"in":[0,1,1026],"p":260}|normal|1
4|Power 4|ZHAPower|Plug 01|OSRAM|7c:b0:3e:aa:00:a3:6b:83-03-0b04|V1.05.09|{"lastupdated":"2018-05-16T10:29:46","power":0}|{"on":true,"reachable":true}|{"d":16,"ep":3,"in":[2820],"p":49246}|normal|1
13|Power 13|ZHAPower|Plug 01|OSRAM|84:18:26:00:00:0a:7c:97-03-0b04|V1.05.09|{"lastupdated":"2018-05-16T10:32:34","power":0}|{"on":true,"reachable":true}|{"d":16,"ep":3,"in":[2820],"p":49246}|normal|1
11|Dimmschalter|ZHASwitch|RWL021|Philips|00:17:88:01:10:5f:6c:94-02-fc00|5.45.1.17846|{"buttonevent":4002,"lastupdated":"2018-05-11T02:08:12"}|{"battery":91,"group":"2135","on":true,"reachable":true}|{"d":2096,"ep":2,"in":[0,1,64512],"out":[6,8,5],"p":49246}|normal|1
2|Power 2|ZHAPower|Plug 01|OSRAM|84:18:26:00:00:0a:48:da-03-0b04|V1.05.09|{"lastupdated":"2018-05-16T10:31:57","power":0}|{"on":true,"reachable":true}|{"d":16,"ep":3,"in":[2820],"p":49246}|normal|1
5|Bewegungsmelder Flur|ZHALightLevel|SML001|Philips|00:17:88:01:02:03:27:b8-02-0400|6.1.0.18912|{"dark":true,"daylight":false,"lastupdated":"2018-05-16T10:31:00","lightlevel":3433,"lux":2}|{"alert":"none","battery":100,"ledindication":false,"on":true,"pending":0,"reachable":true,"usertest":false}|{"d":263,"ep":2,"in":[0,1,1024],"p":260}|normal|1
```

kappes, 2018-05-16:
okay, I played with the firmware version and did some down/upgrade and somehow it is working now.