dresden-elektronik/deconz-rest-plugin #584: Emulating / supporting Hue bridge routines?

dergreg, 2018-05-14:
Hi all

I'm relatively new to deCONZ, but I managed to set up my RaspBee and now have a few Hue lights and Osram Smart Plugs working happily.

Out of curiosity, I tried to add a "routine" (wake up) in the Hue app on the iPhone and enabled `--dbg-http=1` after receiving an error in the app. In the deCONZ logs I noticed that the Hue app tried to create a `CLIPGenericFlag` using the `/sensors` API and received the error message `parameter, recycle, not available`.

Now I understand that `recycle` is not supported according to `rest_sensors.cpp`. But what I wondered was: even if this parameter was supported (even as dummy) and the Hue app managed to add this sensor, what else would be needed to be done/implemented to support Hue app routines?

Anyone else already tried to find this out?

Regards

manup, 2018-05-14:
Good find, adding the recycle, at least as dummy, is easy after that only a test will show what else is needed.

dergreg, 2018-05-16:
Will you add the recycle property to sensors? I would have added it myself and make a PR, but I don’t know of any other changes necessary within deCONZ.

I can do further tests with the Hue app once this is added.

dergreg, 2018-05-21:
Ok, seems like just adding `recycle` to the REST api isn't enough (added and compiled this myself). 

The Hue app complains that there's not enough (disk) space on the bridge to save the routine.
I can only see some HTTP calls to `/api/config` and very few to `/api/capabilities` which does not seem to be supported.

I also hardcoded the api to run in `strict` mode, which didn't have any effect. I guess we're stuck here as we don't know what API output the Hue app actually expects for saving routines.

ebaauw, 2018-05-21:
The Hue bridge doesn’t have a disk - only flash memory.  The Hue app likely queries `/capabilities` to check how many rules or CLIP sensors are still available, and interpretates the error response from deCONZ as: none.

There’s plenty of subtle differences betwen the Hue and deCONZ APIs.  I think it will be challenging to get the Hue app to work with deCONZ in full, especially for Hue lab routines.  To analyse, you could take one of two approaches:
- Sniff the API traffic between the Hue app and deCONZ (or the Hue bridge).  This is regular http traffic so Wireshark will do the trick;
- Dump the content of the Hue bridge (`ph get`), let the Hue app install the routine, dump again, and compare the dumps.

dergreg, 2018-05-28:
I implemented a dummy `/capabilities` endpoint that always returned 1000 of each resources to be available, which has fixed the „no disk space“ error message for at least one routine, another one still complained. Unfortunately I don’t own a Hue bridge to make a dump via `ph get`.

Anyway, I ended up discovering that the Hue app tried calling the `/scenes` api, which likely is related to #29 so I basically gave up at this stage ;-)

closed 2019-03-30 by stale[bot]